carla-simulator/carla #4140: Traffic manager distance for leading vehicles not behaving as expected?

amirx96, 2021-04-24:
** Carla Version: 0.9.11 **
** OS: Ubuntu 18.04 **

Using the following, I thought that the following distance should be bumper to bumper for each vehicle?: 

```python3
self.tm.set_global_distance_to_leading_vehicle(0.0)
```

But the cars still seem to have some distance even if these vehicles are moving or stopped?

E.g.:
[image]

jackbart94, 2021-04-26:
There's always a minimum distance being kept between vehicle, and that parameter is meant to make it higher, not lower. Also, it only shows when the vehicles are stopped. If you want vehicles to ignore collisions between themselves, you can set the ``ignore_vehicles_percentage`` to 100.

amirx96, 2021-04-28:
@jackbart94 Thanks for the clarification. Setting `ignore_vehicles_percentage` to 100 certainly did make vehicles collide - but this happened even when distances were `set_global_distance_to_leading_vehicle(100.0)`

I guess the sort of functionality I am looking for is where the vehicles follow a particular spacing policy. I'll probably have to modify the motion planning code to do that.

closed 2021-04-28 by amirx96