carla-simulator/carla #4141: ‘sensor_tick’ can't change the output rate of radar

Gaopengen, 2021-04-25:
According to the document, 'sensor_tick' can change the simulation seconds between sensor captures, and I tried to change this attribute based on 'maunal_control.py' and monitor the interval of two consecutive radar captures, but it seems not work properly. 

```
class RadarSensor(object):
    def __init__(self, parent_actor):
        self.sensor = None
        self._parent = parent_actor
        self.radar_pcl = queue.Queue()
        world = self._parent.get_world()
        self.debug = world.debug
        self.timestamp = 0
        world = self._parent.get_world()
        bp = world.get_blueprint_library().find('sensor.other.radar')
        bp.set_attribute('horizontal_fov', str(150))
        bp.set_attribute('vertical_fov', str(5))
        bp.set_attribute('points_per_second', str(10000))
        bp.set_attribute('sensor_tick', '0.5')
        bp.set_attribute('range', str(100))
        self.sensor = world.spawn_actor(
            bp,
            carla.Transform(
                carla.Location(x=2.8, z=1.0),
                carla.Rotation()),
            attach_to=self._parent)
        weak_self = weakref.ref(self)
        self.sensor.listen(
            lambda radar_data: RadarSensor._Radar_callback(weak_self, radar_data))

    @staticmethod
    def _Radar_callback(weak_self, radar_data):
        self = weak_self()
        if not self:
            return
        duration = time.time() - self.timestamp
        self.timestamp = time.time()
        print('time duration is:{}s'.format(duration), end='\n')
```
I used asynchronous mode, and I expect to get 2 radar outputs per second(2Hz),  but my output rate is about 50~100Hz. Can anyone help me to figure this out? how can I change radar output rate? Thanks

qqqGpe, 2021-04-25:
I set camera's 'sensor_tick' to 0.25 in the same way, and it works well. So is it a bug for radar?

qqqGpe, 2021-04-26:
This bug has been fixed in the latest version.

closed 2021-04-26 by qqqGpe