carla-simulator/carla #4148: PythonAPI apply_physics_control doesn't work for wheels

namiota, 2021-04-27:
Hello,

I'm using:
- Carla docker version (**carlasim/carla:0.9.11**) 
- **carla_ros_bridge** docker version (built by docker/build.sh script, from master) on **ros:foxy**.

In **carla_ros_bridge** I launched example:
```
ros2 launch carla_ros_bridge carla_ros_bridge_with_example_vehicle.launch.py
```
I'm trying to modify vehicle physics using example taken from: https://carla.readthedocs.io/en/0.9.11/tuto_G_control_vehicle_physics/

Unfortunately, not all parameters are changed in carla. I run separately following script:
```python
import carla

def main():
    # Connect to client
    client = carla.Client('127.0.0.1', 2000)
    client.set_timeout(2.0)

    # Get World and Actors
    world = client.get_world()
    world.wait_for_tick(10)
    all_vehicles = [
        actor for actor in world.get_actors() if "vehicle" in actor.type_id
    ]
    for v in all_vehicles:
        print(v.attributes["role_name"])
    vehicles = [
        vehicle
        for vehicle in all_vehicles
        if vehicle.attributes["role_name"] == "ego_vehicle"
    ]
    # Get first ego_vehicle
    vehicle = vehicles[0]

    # Create Wheels Physics Control
    front_left_wheel  = carla.WheelPhysicsControl(tire_friction=4.5, damping_rate=1.0, max_steer_angle=70.0, radius=30.0)
    front_right_wheel = carla.WheelPhysicsControl(tire_friction=2.5, damping_rate=1.5, max_steer_angle=70.0, radius=25.0)
    rear_left_wheel   = carla.WheelPhysicsControl(tire_friction=1.0, damping_rate=0.2, max_steer_angle=70.0,  radius=15.0)
    rear_right_wheel  = carla.WheelPhysicsControl(tire_friction=1.5, damping_rate=1.3, max_steer_angle=70.0,  radius=20.0)

    wheels = [front_left_wheel, front_right_wheel, rear_left_wheel, rear_right_wheel]

    # Change Vehicle Physics Control parameters of the vehicle
    physics_control = vehicle.get_physics_control()

    physics_control.torque_curve = [carla.Vector2D(x=0, y=400), carla.Vector2D(x=1300, y=600)]
    physics_control.max_rpm = 10000
    physics_control.moi = 1.0
    physics_control.damping_rate_full_throttle = 0.0
    physics_control.use_gear_autobox = True
    physics_control.gear_switch_time = 0.5
    physics_control.clutch_strength = 10
    physics_control.mass = 10000
    physics_control.drag_coefficient = 0.25
    physics_control.steering_curve = [carla.Vector2D(x=0, y=1), carla.Vector2D(x=100, y=1), carla.Vector2D(x=300, y=1)]
    physics_control.wheels = wheels
    physics_control.use_sweep_wheel_collision = True

    print("FIRST")
    print(physics_control)

    # Apply Vehicle Physics Control for the vehicle
    vehicle.apply_physics_control(physics_control)

    world = client.get_world()
    world.wait_for_tick(10)

    physics_control = vehicle.get_physics_control()
    print("SECOND")
    print(physics_control)

if __name__ == '__main__':
    main()
```
And got results:
```python
FIRST
VehiclePhysicsControl(
  torque_curve=[
    Vector2D(x=0.000000, y=400.000000), 
    Vector2D(x=1300.000000, y=600.000000)
  ], 
  max_rpm=10000.000000, 
  moi=1.000000, 
  damping_rate_full_throttle=0.000000, 
  damping_rate_zero_throttle_clutch_engaged=2.000000, 
  damping_rate_zero_throttle_clutch_disengaged=0.350000, 
  use_gear_autobox=True, 
  gear_switch_time=0.500000, 
  clutch_strength=10.000000, 
  final_ratio=4.000000, 
  forward_gears=[
    GearPhysicsControl(ratio=3.590000, down_ratio=0.200000, up_ratio=0.350000), 
    GearPhysicsControl(ratio=2.190000, down_ratio=0.200000, up_ratio=0.350000), 
    GearPhysicsControl(ratio=1.410000, down_ratio=0.200000, up_ratio=0.350000), 
    GearPhysicsControl(ratio=1.000000, down_ratio=0.200000, up_ratio=0.350000), 
    GearPhysicsControl(ratio=0.830000, down_ratio=0.200000, up_ratio=0.350000)
  ], 
  mass=10000.000000, 
  drag_coefficient=0.250000, 
  center_of_mass=Location(x=0.100000, y=0.000000, z=-0.200000), 
  steering_curve=[
    Vector2D(x=0.000000, y=1.000000), 
    Vector2D(x=100.000000, y=1.000000), 
    Vector2D(x=300.000000, y=1.000000)
  ], 
  wheels=[
    WheelPhysicsControl(
      tire_friction=4.500000, 
      damping_rate=1.000000, 
      max_steer_angle=70.000000, 
      radius=30.000000, 
      max_brake_torque=1500.000000, 
      max_handbrake_torque=3000.000000, 
      position=Vector3D(x=0.000000, y=0.000000, z=0.000000)
    ), 
    WheelPhysicsControl(
      tire_friction=2.500000, 
      damping_rate=1.500000, 
      max_steer_angle=70.000000, 
      radius=25.000000, 
      max_brake_torque=1500.000000, 
      max_handbrake_torque=3000.000000, 
      position=Vector3D(x=0.000000, y=0.000000, z=0.000000)
    ), 
    WheelPhysicsControl(
      tire_friction=1.000000, 
      damping_rate=0.200000, 
      max_steer_angle=70.000000, 
      radius=15.000000, 
      max_brake_torque=1500.000000, 
      max_handbrake_torque=3000.000000, 
      position=Vector3D(x=0.000000, y=0.000000, z=0.000000)
    ), 
    WheelPhysicsControl(
      tire_friction=1.500000, 
      damping_rate=1.300000, 
      max_steer_angle=70.000000, 
      radius=20.000000, 
      max_brake_torque=1500.000000, 
      max_handbrake_torque=3000.000000, 
      position=Vector3D(x=0.000000, y=0.000000, z=0.000000)
    )
  ], 
  use_sweep_wheel_collision=1)

SECOND
VehiclePhysicsControl(
  torque_curve=[
    Vector2D(x=0.000000, y=400.000000), 
    Vector2D(x=1300.000000, y=600.000000)
  ], 
  max_rpm=10000.000000, 
  moi=1.000000, 
  damping_rate_full_throttle=0.000000, 
  damping_rate_zero_throttle_clutch_engaged=2.000000, 
  damping_rate_zero_throttle_clutch_disengaged=0.350000, 
  use_gear_autobox=True, 
  gear_switch_time=0.500000, 
  clutch_strength=10.000000, 
  final_ratio=4.000000, 
  forward_gears=[
    GearPhysicsControl(ratio=3.590000, down_ratio=0.200000, up_ratio=0.350000), 
    GearPhysicsControl(ratio=2.190000, down_ratio=0.200000, up_ratio=0.350000), 
    GearPhysicsControl(ratio=1.410000, down_ratio=0.200000, up_ratio=0.350000), 
    GearPhysicsControl(ratio=1.000000, down_ratio=0.200000, up_ratio=0.350000), 
    GearPhysicsControl(ratio=0.830000, down_ratio=0.200000, up_ratio=0.350000)
  ], 
  mass=10000.000000, 
  drag_coefficient=0.250000, 
  center_of_mass=Location(x=0.100000, y=0.000000, z=-0.200000), 
  steering_curve=[
    Vector2D(x=0.000000, y=1.000000), 
    Vector2D(x=100.000000, y=1.000000), 
    Vector2D(x=300.000000, y=1.000000)
  ], 
  wheels=[
    WheelPhysicsControl(
      tire_friction=1.500000, 
      damping_rate=0.250000, 
      max_steer_angle=69.999992, 
      radius=35.000000, 
      max_brake_torque=1500.000000, 
      max_handbrake_torque=0.000000, 
      position=Vector3D(x=39337.871094, y=16553.234375, z=34.950100)
    ), 
    WheelPhysicsControl(
      tire_friction=1.500000, 
      damping_rate=0.250000, 
      max_steer_angle=69.999992, 
      radius=35.000000, 
      max_brake_torque=1500.000000, 
      max_handbrake_torque=0.000000, 
      position=Vector3D(x=39156.066406, y=16553.453125, z=34.950222)
    ), 
    WheelPhysicsControl(
      tire_friction=1.500000, 
      damping_rate=0.250000, 
      max_steer_angle=0.000000, 
      radius=35.000000, 
      max_brake_torque=1500.000000, 
      max_handbrake_torque=3000.000000, 
      position=Vector3D(x=39339.367188, y=16248.850586, z=34.950108)
    ), 
    WheelPhysicsControl(
      tire_friction=1.500000, 
      damping_rate=0.250000, 
      max_steer_angle=0.000000, 
      radius=35.000000, 
      max_brake_torque=1500.000000, 
      max_handbrake_torque=3000.000000, 
      position=Vector3D(x=39154.570312, y=16249.069336, z=34.950241)
    )
  ], 
  use_sweep_wheel_collision=0)
```
As can be seen from the above example, wheel physics were not applied.

I've checked above example in **carlasim/carla:0.9.10** and it works fine. Am I doing something wrong?

Thanks for help in advance.

jbenzhhn, 2021-05-14:
I have a problem with the wheel collisions too. When I add a new vehicle and use an own shape as wheel_colission there is no visible effect. Maybe it's related to the same issue, cause in manual_control the variable use_sweep_wheel_colission is set to true.

joel-mb, 2021-05-31:
Hi @namiota. Sorry for the delay. This is a known issue in `0.9.11` and it has been fixed in `dev` (PR: https://github.com/carla-simulator/carla/pull/4128). We have also added new parameters to control the stiffness of the wheels. You can take a look here: https://carla.readthedocs.io/en/latest/python_api/#carla.WheelPhysicsControl

@jbenzhhn which CARLA version are you using?

namiota, 2021-06-02:
Thanks @joel-mb for the answer. 
When new version of the CarlaSim will be released?

jbenzhhn, 2021-06-02:
I´m using Carla 0.9.11.

ElektronLibre, 2021-06-11:
Hi guys,

I´m using Carla 0.9.11 too and i have the same problem, all parameters are not applied, As can be seen from the namiota's example above, **the max_steer_angle of front and rear wheel is not modified too**.
For a research project, i wish to change the max steer angle of front and rear wheels at runtime.

there is someone have a solution? except using the latest version

hee0624, 2022-02-17:
> there is someone have a solution? except using the latest version

i'm using Carla 0.9.12 , and i have the same problem.wheels parameter is not applied.

redbananamoon, 2022-02-24:
Same problem with 0.9.13 (running in synchronous mode), clearly the issue has not been fixed.

redbananamoon, 2022-02-28:
In addition to that, CARLA crashes if the wheel radius (for example) is modified in the blueprint's wheel physics in UE editor, after spawning an ego vehicle that has it's wheel radius modified.