carla-simulator/carla #4149: Radial velocities of radar are not correct

qqqGpe, 2021-04-27:
Version: Carla nightly version (donwload on 26 Apr. 2021)
Platform: Windows 10
Python: 3.7

Hi, I want to sample some radar point cloud and radial velocities for my experiment, but I find that the radial velocities I got is odd. Here is my code snippet:
`
class RadarSensor(object):
    def __init__(self, parent_actor):
        self.sensor = None
        self._parent = parent_actor
        self.velocity_range = 7.5 # m/s
        self.frame_count = -1
        world = self._parent.get_world()
        self.debug = world.debug
        self.pre_location = carla.Location()
        self.pre_time = time.time()
        bp = world.get_blueprint_library().find('sensor.other.radar')
        bp.set_attribute('horizontal_fov', str(150))
        bp.set_attribute('vertical_fov', str(3))
        bp.set_attribute('points_per_second', str(20000))
        bp.set_attribute('sensor_tick', str(0.33))
        bp.set_attribute('range', str(100))
        self.sensor = world.spawn_actor(
            bp,
            carla.Transform(
                carla.Location(x=2.8, z=3.0),
                carla.Rotation()),
            attach_to=self._parent)
        # We need a weak reference to self to avoid circular reference.
        weak_self = weakref.ref(self)
        self.sensor.listen(
            lambda radar_data: RadarSensor._Radar_callback(weak_self, radar_data))

    @staticmethod
    def _Radar_callback(weak_self, radar_data):
        self = weak_self()
        if not self:
            return

        self.frame_count = self.frame_count + 1
        current_rot = radar_data.transform.rotation
        current_loc = radar_data.transform.location
        current_time = time.time()
        delta_t = current_time - self.pre_time
        delta_p = current_loc - self.pre_location

        T = np.array(carla.Transform(
            current_loc,
            carla.Rotation(
                pitch=current_rot.pitch,
                yaw=current_rot.yaw,
                roll=current_rot.roll)).get_matrix())

        delta_p = T[:3,:3].transpose().dot(np.array((delta_p.x, delta_p.y, delta_p.z)))
        self.pre_location = current_loc
        self.pre_time = current_time
        print('delta t is:%.3f s, delta p is:%.3f, %.3f, %.3f, vel is:%.3f, %.3f, %.3f'
              %(delta_t,
                delta_p[0], delta_p[1], delta_p[2],
                delta_p[0]/delta_t, delta_p[1]/delta_t, delta_p[2]/delta_t))

        pcl = list()
        for detect in radar_data:
            azi = math.degrees(detect.azimuth)
            alt = math.degrees(detect.altitude)
            vec = carla.Vector3D(x=detect.depth)
            vel = carla.Vector3D(x=detect.velocity)

            carla.Transform(
                carla.Location(),
                carla.Rotation(
                    pitch=alt,
                    yaw=azi,
                    roll=current_rot.roll)).transform(vec)

            carla.Transform(
                carla.Location(),
                carla.Rotation(
                    pitch=alt,
                    yaw=azi,
                    roll=current_rot.roll)).transform(vel)
            pcl.append([vec.x, vec.y, vec.z, vel.x, vel.y, vel.z, detect.velocity])

        ###########
        # save data
        ###########
        fpath = './_out/exp/pcl_%d.csv' % self.frame_count
        np.savetxt(fpath, np.array(pcl), delimiter=',')
`
a console output snippet is showed below, it looks quite normal.
`
... ...
delta t is:0.310 s, delta p is:9.752, -0.039, -0.077 m, vel is:31.441, -0.125, -0.248 m/s
delta t is:0.344 s, delta p is:10.879, 0.408, -0.081 m, vel is:31.598, 1.184, -0.235 m/s
delta t is:0.322 s, delta p is:10.297, 0.386, -0.077 m, vel is:31.964, 1.197, -0.239 m/s
delta t is:0.324 s, delta p is:10.404, 0.261, -0.079 m, vel is:32.077, 0.804, -0.244 m/s
delta t is:0.341 s, delta p is:10.963, 0.965, -0.059 m, vel is:32.144, 2.829, -0.173 m/s
delta t is:0.325 s, delta p is:10.415, 0.664, -0.030 m, vel is:32.003, 2.040, -0.093 m/s
delta t is:0.341 s, delta p is:11.280, 0.245, -0.039 m, vel is:33.070, 0.718, -0.116 m/s
delta t is:0.319 s, delta p is:10.469, 0.654, -0.053 m, vel is:32.804, 2.050, -0.166 m/s
... ...
`
But the data I saved to csv file looks weird, the radial velocity is too fast.
`
... ...
       x |        y |        z |       vx |         vy |     vz | radial_velocity
10.33 | 20.89 | -0.17 | -62.34 | -126.05 | 1.03 | -140.63
75.37 | -99.63 | 0.92 | -115.59 | 152.80 | -1.42 | -191.60
10.74 | 20.84 | -0.22 | -66.51 | -129.14 | 1.34 | -145.27
34.48 | -94.74 | -0.46 | -37.17 | 102.13 | 0.49 | -108.69
30.25 | 25.64 | 0.55 | -184.65 | -156.48 | -3.34 | -242.06
7.71 | 21.18 | -0.02 | -37.10 | -101.93 | 0.08 | -108.47
67.26 | -40.34 | -0.07 | -233.53 | 140.06 | 0.25 | -272.31
87.15 | 26.31 | -0.24 | -290.91 | -87.82 | 0.81 | -303.88
45.39 | -95.68 | 0.20 | -58.36 | 123.03 | -0.25 | -136.17
53.15 | 36.30 | -0.01 | -216.42 | -147.81 | 0.05 | -262.08
36.21 | -94.88 | -0.41 | -40.41 | 105.87 | 0.46 | -113.32
27.08 | 25.93 | 0.06 | -165.54 | -158.52 | -0.35 | -229.19
11.47 | 20.76 | 0.04 | -74.16 | -134.26 | -0.23 | -153.38
... ...
`
There is no possible radial velocities can exceed 100m/s. So, is it a bug. or i made some mistakes? Can anyone help me to solve this problem?

qqqGpe, 2021-04-28:
@doterop Could you please help me to figure this out?

DSantosO, 2021-05-12:
Hello @qqqGpe,
There is a couple of things that I don't see correct in your computations. As I imagine that you are in asynchronous mode, be very careful about what you compare. The client is going to receive the callback of the radar at some point after is being computed so don't use `time.time()` to compute the time. That has nothing to do with the simulation time and it is being saved when the callback is executed in the client. Use `radar_data.timestamp` to get the simulation timestamp at which the data were computed. Second, be careful also with computing the velocity of the points subtracting with the previous points. You cannot be sure that the N detection of two frames will correspond to the same location in the world. 

Anyway, these two things have nothing to do with the detection velocities. I have run a test with a car driving at a constant velocity of 60 km/h (around 16.6 m/s) and then checked the detections. As the car is the only dynamic object all detections should be more or less below that value. I also removed the `bp.set_attribute('sensor_tick', str(0.33))` option for the test.

This is what I get:
```
delta t is:0.151 s, Car: 15.020 Detection0: -10.441 Aver: -10.365 Max: -10.575 Min: -9.769
delta t is:0.134 s, Car: 15.787 Detection0: -11.873 Aver: -11.671 Max: -11.904 Min: -11.005
delta t is:0.072 s, Car: 15.660 Detection0: -16.196 Aver: -15.879 Max: -16.196 Min: -14.967
delta t is:0.060 s, Car: 15.931 Detection0: -16.307 Aver: -15.984 Max: -16.309 Min: -15.076
delta t is:0.067 s, Car: 16.006 Detection0: -16.219 Aver: -15.917 Max: -16.239 Min: -15.017
delta t is:0.063 s, Car: 14.996 Detection0: -15.541 Aver: -15.969 Max: -16.281 Min: -15.041
delta t is:0.126 s, Car: 15.024 Detection0: -12.513 Aver: -12.339 Max: -12.584 Min: -11.633
```
I am comparing the speed of the car, the radial velocity of detection 0, and then the average, the maximum, and the minimum velocity.
Notice that the velocity of the car does not necessarily correspond with the frame at which the radar data was computed but as it is supposed to be more or less constant is should be fine.

The problem appears if the `sensor_tick` is too high. For the value of `0.33` I also have observed values that make no sense. It seems that we have a problem with our computation of the velocity in this case. I hope you can use a lower value (or remove the attribute to get the data for all ticks) and we will try to fix it as soon as we can.

qqqGpe, 2021-05-13:
Thanks for your suggestion, I'm also looking forward to the new iteration.

closed 2023-06-02 by stale[bot]