carla-simulator/carla #4151: Orthographic projection

annaornatskaya, 2021-04-27:
Hello!
I added the choice of the projection type to CARLA's RGB camera but run into a problem: when I choose the orthographic projection the image is shifting at every tick, while camera is in the same location. But when I choose the perspective projection and then I change the type to orthographic in UE4 Editor (after beginning of the simulation), everything is ok.
Maybe someone knows something about the problem?

DSantosO, 2021-05-12:
Hello @annaornatskaya,
What do you mean that the image is shifting? As if the camera were moving in one direction? 
Also, can you give us more details on how you add the option for the orthographic projection in the RGB Camera?

annaornatskaya, 2021-05-12:
Hello @DSantosO, 
> What do you mean that the image is shifting? As if the camera were moving in one direction?

No, it's more like as if the camera is moving in a circle in different directions. I made a screen record to show (shifting/not shifting after changing the projection type). 

[image]

> Also, can you give us more details on how you add the option for the orthographic projection in the RGB Camera?

````
void ASceneCaptureSensor::SetCamType(bool CamType)
{
    check(CaptureComponent2D != nullptr);
    if (CamType)
    {
      CaptureComponent2D->ProjectionType = ECameraProjectionMode::Type::Perspective;
    }
    else
    {
      CaptureComponent2D->ProjectionType = ECameraProjectionMode::Type::Orthographic;
    }
} 
````

CamType - rgb camera attribute

DSantosO, 2021-05-26:
Hello @annaornatskaya,
Sorry for the late response, I was trying to understand what was happening.
The same was happening to me when I tried to do the same: 
https://user-images.githubusercontent.com/5548733/119621658-adce6d00-be06-11eb-9121-ee7708703544.mp4

I was checking the source of unreal and when an option is changed in the editor there are lines to update the camera that only happened in the editor (at Engine\Private\Components\SceneCaptureComponent.cpp), for example, to update the frustum. 
Anyway, I tried to include those on our camera but I could not make it work. You can also check the CameraComponent (the one used by the spectator) to check if doing the change there works correctly because it can give you a hint of why is not working in the SceneCaptureComponent.

annaornatskaya, 2021-05-30:
Hello @DSantosO,

It looks like the solution is very simple:

````
void ASceneCaptureSensor::SetCamType(bool CamType)
{
    check(CaptureComponent2D != nullptr);
    if (CamType)
    {
      CaptureComponent2D->ProjectionType = ECameraProjectionMode::Perspective;
    }
    else
    {
      CaptureComponent2D->ProjectionType = ECameraProjectionMode::Orthographic;
    }
}
````
So now there's no shifting. 

[image]

But unfortunately, one problem remains: the camera doesn't see the sky.

[image]

dmitry0000, 2021-06-21:
Looking at this "jumpy" behavior of camera orientation (however, not completely random), this may be caused by TAA, which behaves this strange way for the case of orthographic projection.....

annaornatskaya, 2021-06-21:
> Looking at this "jumpy" behavior of camera orientation (however, not completely random), this may be caused by TAA, which behaves this strange way for the case of orthographic projection..... I.e. the TAA displacements are too big for the latter... Probably, can be fixed somewhere in UE4 code, but for now just use different type of AA

This is it, TAA was set in my project. I tried to launch with FXAA and without any AA, and the camera works for both! 
Thank you)

closed 2023-06-02 by stale[bot]