carla-simulator/carla #4154: what is the default fps value for carla server

nuomizai, 2021-04-28:
Does anyone know what is the default fps value for CARLA server? I didn't find that in the document.

corkyw10, 2021-05-12:
Hi @nuomizai, there isn't really a default fps per say. The default for carla is to run in asynchronous mode with a variable time step, in that case the simulation speed will be driven by the power of your machine. There's more info about it [here](https://carla.readthedocs.io/en/0.9.11/adv_synchrony_timestep/#possible-configurations)

nuomizai, 2021-05-15:
Thanks for your reply, @corkyw10 . So, does it mean that there is no need to set the server from the terminal, while I can just pass the settings to the server through client by `apply_settings`?

corkyw10, 2021-05-17:
@nuomizai, that's right