carla-simulator/carla #4155: spawn walkers error with python exit code 139  (interrupted by signal 11: sigsegv)

lish0, 2021-04-28:
I tried to spawn walkers with spawn_npc.py, but found the process died at world.get_random_location_from_navigation(). Do I misconfigure something? I use python3.9.4 as interpreter.

arihant2math, 2021-04-30:
Use python 3.7 instead, see #4087

lish0, 2021-05-03:
> Use python 3.7 instead, see #4087 

Thanks for your reply. But I was able to install .egg file and import carla, In my carla workspace, vehicles can be successfully spawned while the pedestrian can't.

arihant2math, 2021-05-03:
What is is run command, is it `python3 spawn_npc.py` (`python spawn_npc.py` for windows users)?

corkyw10, 2021-05-12:
Hi @lish0, are you using a custom map or one of the CARLA maps? What version of CARLA are you using? What output do you get with `python3 spawn_npc.py -n 0 -w 10`?

lish0, 2021-05-14:
> Hi @lish0, are you using a custom map or one of the CARLA maps? What version of CARLA are you using? What output do you get with `python3 spawn_npc.py -n 0 -w 10`?
> 
> 

Hi @corkyw10, I use official carla map, and the CARLA version is 0.9.11. When I try to spawn walkers, I get the only output is "python exit code 139  (interrupted by signal 11: sigsegv)", and   the UE4 editor gets stuck at the same time.

closed 2023-06-02 by stale[bot]