carla-simulator/carla #4173: get_waypoint does not return closest waypoint

syveqc, 2021-05-05:
Hello,

we recently discovered that `map.get_waypoint(pos)` does not return the closest waypoint to the given location sometimes, see the following test script:
```
import carla

def norm(vec):
    return (vec.x**2+vec.y**2+vec.z**2)**0.5

def get_waypoint(map, pos):
    wp = map.get_waypoint(pos)
    wp = wp.previous(1)[0]
    other_wp = wp
    dist = norm(pos-wp.transform.location)
    for _ in range(40):
        other_wp = other_wp.next(1/20)[0]
        other_dist = norm(pos-other_wp.transform.location)
        if other_dist < dist:
            wp = other_wp
            dist = other_dist
    return wp

client = carla.Client("localhost", 2000)
world = client.load_world("Town04")
map = world.get_map()
def test_waypoint(x,y,z):
    pos = carla.Location(x=x, y=y, z=z)

    wp = map.get_waypoint(pos)
    other_wp = get_waypoint(map, pos)

    print(f"distance of wp to pos      : {norm(wp.transform.location-pos)}")
    print(f"distance of other_wp to pos: {norm(other_wp.transform.location-pos)}")
    print()

test_waypoint(-480, 25, 0.5)
test_waypoint(-472.43, 356.81, 0.0)
test_waypoint(-501.42, 302.58, 0.0)
```

By searching nearby waypoints for closer ones, we get quite some distance between this waypoint and the one `map.get_waypoint` returns. The output of the script is
```
distance of wp to pos      : 0.9172027597854976
distance of other_wp to pos: 0.7628350928170309

distance of wp to pos      : 0.20053650461407116
distance of other_wp to pos: 0.0710354893770711

distance of wp to pos      : 0.3470711781046071
distance of other_wp to pos: 0.03117576364531771
```

We currently use this method to get a vehicles offset from the center of the road, so this large discrepancy can be a bit problematic.

hh0rva1h, 2021-08-19:
Go away stale bot.

hh0rva1h, 2022-02-16:
@Axel1092 Please do actually have a look before stale bot closes your issues. This shouldn't take too much time, since a code example is given.

hh0rva1h, 2022-04-29:
@Axel1092 Demonstration code is given, please check and give feedback whether this is actionable before stale blot closes this.

M-Colley, 2026-10-26:
Draft PR #10051 proposes the `ue58-dev` correction for the three reported Town04 queries: convert indexed lane-chord distance into reference-line s, and keep curved/changing lane profiles out of the straight-lane shortcut. The genuine Town04 OpenDrive fixture reproduces the reported behavior. All 16 regressions pass on Windows and Linux, with Clang sanitizer and existing-native integration checks also passing. The existing sampled-segment index is retained, so this is not a claim of exact global curve minimization; live simulator/autopilot validation remains unrun.