carla-simulator/carla #4177: Software requirement error when building CARLA from source on Linux

AIasd, 2021-05-08:
Hi,

I am following [this instruction](https://carla.readthedocs.io/en/latest/build_linux/) to build CARLA (master branch) on Ubuntu18.04.

When I run the suggested commands to satisfy software requirements
```
sudo apt-get update &&
sudo apt-get install wget software-properties-common &&
sudo add-apt-repository ppa:ubuntu-toolchain-r/test &&
wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key|sudo apt-key add - &&
sudo apt-add-repository "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" &&
sudo apt-get update
```
I encountered the following error:

```
Get:1 file:/var/cuda-repo-ubuntu1804-11-1-local  InRelease
Ign:1 file:/var/cuda-repo-ubuntu1804-11-1-local  InRelease
Get:2 file:/var/cuda-repo-ubuntu1804-11-1-local  Release [564 B]
Hit:4 http://packages.ros.org/ros/ubuntu bionic InRelease
Hit:5 http://linux.teamviewer.com/deb stable InRelease
Get:2 file:/var/cuda-repo-ubuntu1804-11-1-local  Release [564 B]                                                                 
Ign:6 http://dist.carla.org/carla-0.9.7 bionic InRelease                                                                         
Hit:7 http://dist.carla.org/carla-ros-bridge-melodic bionic InRelease                                                            
Hit:8 http://dist.carla.org/carla-0.9.8 all InRelease                                                                            
Hit:9 https://nvidia.github.io/libnvidia-container/ubuntu18.04/amd64  InRelease                                                  
Hit:10 https://nvidia.github.io/nvidia-container-runtime/ubuntu18.04/amd64  InRelease                                            
Hit:11 https://download.docker.com/linux/ubuntu bionic InRelease                                                                 
Hit:12 https://nvidia.github.io/nvidia-docker/ubuntu18.04/amd64  InRelease                                                       
Err:13 http://dist.carla.org/carla-0.9.7 bionic Release                                                                          
  404  Not Found [IP: 34.227.255.250 80]
Hit:14 https://cloud.r-project.org/bin/linux/ubuntu bionic-cran35/ InRelease                                                     
Hit:15 https://packages.microsoft.com/repos/ms-teams stable InRelease                                                            
Hit:3 https://apt.llvm.org/xenial llvm-toolchain-xenial-8 InRelease                                                              
Ign:16 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  InRelease                                     
Hit:17 http://us.archive.ubuntu.com/ubuntu bionic InRelease                                                                      
Hit:18 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release                                       
Get:19 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]                                                     
Get:20 http://us.archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]                                                    
Get:21 http://us.archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]                                                  
Hit:22 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease                                                     
Ign:23 http://ppa.launchpad.net/mc3man/trusty-media/ubuntu bionic InRelease                       
Hit:25 http://ppa.launchpad.net/obsproject/obs-studio/ubuntu bionic InRelease                     
Hit:26 http://ppa.launchpad.net/sylvain-pineau/kazam/ubuntu bionic InRelease                      
Hit:24 https://packagecloud.io/github/git-lfs/ubuntu bionic InRelease                             
Hit:28 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu bionic InRelease
Err:29 http://ppa.launchpad.net/mc3man/trusty-media/ubuntu bionic Release
  404  Not Found [IP: 91.189.95.85 80]
Reading package lists... Done 
E: The repository 'http://dist.carla.org/carla-0.9.7 bionic Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://ppa.launchpad.net/mc3man/trusty-media/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Skipping acquire of configured file 'main/binary-i386/Packages' as repository 'http://dist.carla.org/carla-ros-bridge-melodic bionic InRelease' doesn't support architecture 'i386'
```

I guess because of the above error, when I proceed to the command
```
make PythonAPI
```
to compile the CARLA client, I encountered the following error:

```
Setup.sh: Extracting boost for Python 3.
Building Boost.Build engine with toolset clang... tools/build/src/engine/b2
Unicode/ICU support for Boost.Regex?... /usr
Generating Boost.Build configuration in project-config.jam for clang...

Bootstrapping is done. To build, run:

    ./b2
    
To generate header files, run:

    ./b2 headers

To adjust configuration, edit 'project-config.jam'.
Further information:

   - Command line help:
     ./b2 --help
     
   - Getting started guide: 
     http://www.boost.org/more/getting_started/unix-variants.html
     
   - Boost.Build documentation:
     http://www.boost.org/build/

sh: 1: clang++: not found
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/tools/clang-linux.jam:63: in clang-linux.init
*** argument error
* rule clang.init-cxxstd-flags ( toolset : condition * : version )
* called with: ( clang-linux : <toolset>clang-linux :  )
* missing argument version
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/tools/clang.jam:54:see definition of rule 'clang.init-cxxstd-flags' being called
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/build/toolset.jam:44: in toolset.using
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/tools/clang.jam:33: in clang.init
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/build/toolset.jam:44: in toolset.using
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/build/project.jam:1104: in using
project-config.jam:12: in modules.load
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/build-system.jam:255: in load-config
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/build-system.jam:486: in load-configuration-files
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/build-system.jam:607: in load
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/kernel/modules.jam:295: in import
/mnt/sda/carla/Build/boost-1.72.0-c8-source/tools/build/src/kernel/bootstrap.jam:139: in boost-build
/mnt/sda/carla/Build/boost-1.72.0-c8-source/boost-build.jam:17: in module scope
Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed
make: *** [setup] Error 1
```

Can someone sheds some light on possible ways to get around this error? Thanks a lot!

corkyw10, 2021-05-13:
Hi @AIasd, I am not sure about the installation of the software requirements. But if you are installing UE4.24 and CARLA master, then you may be using the wrong documentation. The one you linked is the "latest" version of the docs, these reference the development branch of CARLA which uses UE4.26. Select the correct version of the docs (0.9.11 for the current master) to get the correct documentation. Try again with a fresh installation.

AIasd, 2021-05-13:
Hi @corkyw10 , thank you for the help! I can now build it successfully.

closed 2021-05-13 by AIasd