carla-simulator/carla #4185: Scenario Runner: watchdog timeout error

NikhilKalghatgi, 2021-05-10:
I am trying to execute a scenario using scenario_runner , however when i run it i am getting this error at random frames:
**Watchdog exception - Timeout of 11.0 seconds occurred** 

I found out that the server gets stuck after calling .tick() from scenario_manager.py:
**CarlaDataProvider.get_world().tick(10)**
As you can see i've given timeout parameter as 10 seconds and yet i'm seeing this error.

I've integrated my AD client in external_control.py in the run_step() function. I hope the details are enough.
I am using carla version 0.9.10 and also tried with 0.9.11 but the issue still exists.
System configuration: Intel i7-9th gen, RTX2080 8gb, 32gb RAM, OS: Ubuntu 18.04

Note: The timeout is random, sometimes the execution completes successfully without this timeout error.
Any help or suggestion would be appreciated, thank you.

glopezdiest, 2021-05-14:
Hey @NikhilKalghatgi, if the CARLA tick is causing a timeout, then the problem is server related. What happens to the server when that timeout happens. Is it frozen, shutdown, or anything of the sorts? If that is the case, as you said, the client will get stuck when trying to tick it.

NikhilKalghatgi, 2021-05-14:
@glopezdiest  Thank you for the response. Yeah the server gets frozen, wanted to understand why that happens. Is there any way to avoid it ?

glopezdiest, 2021-05-17:
You said you were using CARLA 0.9.10 and 0.9.11 but what version of ScenarioRunner are you using? What commands are you trying to execute exactly?

NikhilKalghatgi, 2021-05-17:
I am scenarioRunner version 0.9.11 . I am running scenario_runner.py with open scenario parser.

NikhilKalghatgi, 2021-05-24:
@glopezdiest Hi, on further digging I found that there's a deadlock in traffic manager tick in TrafficManagerLocal.cpp.
**step_end_trigger.wait(lock, [this]() { return step_end.load(); });**
The above line is where its getting stuck.

glopezdiest, 2021-05-25:
Hey @NikhilKalghatgi. thanks for the input. I actually havent't seen that issue, which seems weird, but we'll investigate it regardless. tagging @jackbart94 too, he'll know more about the TM

NikhilKalghatgi, 2021-05-25:
> Hey @NikhilKalghatgi. thanks for the input. I actually havent't seen that issue, which seems weird, but we'll investigate it regardless. tagging @jackbart94 too, he'll know more about the TM. Just to recap, you are using CARLA and SR 0.9.11, and are using the `python scenario_runner.py --openscenario` command?

Yup @glopezdiest that's right.

closed 2023-06-02 by stale[bot]