carla-simulator/carla #4188: Carla-client error while trying to connect to server running in local docker container

dhiegomaga, 2021-05-10:
Doing:

`sudo nvidia-docker run -it --rm  carlasim/carla:0.9.10 /bin/bash`

and then:

`SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl`

I get these results: 

4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default

But maybe this is normal? When checking `nvidia-smi` it does appear to be using the card. However, I am unable to connect to the carla-server using a local client... 

> Traceback (most recent call last):
>   File "set_environment.py", line 100, in main
>     world = client.load_world(args.town)
> RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

_Originally posted by @dhiegomaga in https://github.com/carla-simulator/carla/issues/3431#issuecomment-836657242_

dhiegomaga, 2021-05-10:
I found the problem. It was discussed [here](https://github.com/carla-simulator/carla/issues/3476). 

When running the docker, the port should be specified: 

```
sudo nvidia-docker run -p 2000-2002:2000-2002 -it --rm  carlasim/carla:0.9.10 /bin/bash
```

Then carla can be started headless on the docker, and accessible on port 2000: 

```bash
SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl
```

corkyw10, 2021-05-13:
@dhiegomaga, glad you got it working!

closed 2021-05-13 by corkyw10