carla-simulator/carla #4189: Problem with bounding boxes of 2 wheels vehicles

guygo12345, 2021-05-10:
Hey!

I'm trying to get the bounding boxes of all the vehicles in a frame.
I'm iterating over the vehicle type actors and extract their bounding boxes with actor.bounding_box API
This works pretty much good, but in some cases I get wrong bounding boxes specifically on bicycles and bikes.
The result is that zero width boxes, with wrong location/rotation. It's even more weird because I do get the right boxes on different actors, including some of the bicycles and bikes

Would love to get some help here.. 

Thanks!

guygo12345, 2021-06-13:
Any news about that?

beratersari, 2022-09-29:
Hey! How did you get the frames? I can reach only log file which has the x y z coordinates of the objects. Could help me? Thanks in advance!

zhuzhouzhou, 2023-06-01:
this issue still excited with 0.9.14

Bo-Yu-Columbia, 2023-07-20:
The issue still exists with 0.9.14