carla-simulator/carla #4192: Document API calls in the examples

qhaas, 2021-05-11:
The [`set_pedestrians_cross_factor` function of `carla.World` is invoked in the official 0.9.11 pythonapi examples](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/examples/spawn_npc.py#L258).  But, it does not appear in the [documentation](https://carla.readthedocs.io/en/0.9.11/python_api/#methods_58).  It would be useful to developers to document any API calls used in the official examples in the official API documentation so we can easily understand the behavior, nuances, defaults, etc. associated with said API calls.

I have not checked all the API calls in the examples to see if they are documented, so there might be more.

corkyw10, 2021-05-17:
Hi @qhass, thank you for the input. I will review the examples and update the Python API as required.

corkyw10, 2021-08-05:
You can find this method documented [here](https://carla.readthedocs.io/en/0.9.12/python_api/#carla.World.set_pedestrians_cross_factor)

closed 2021-08-05 by corkyw10