carla-simulator/carla #4196: Get current RPM of vehicle

HarishKumar051993, 2021-05-12:
Hi, I'm trying to find a way to get the current revolutions per minute (RPM) of any wheel at a given time via the Python API.
I know I can get the throttle level and the max RPM of the engine via VehiclePhysicsControl, so in theory I could just derive the wheel RPM from there. Due to gearbox properties etc. that value will be pretty inaccurate I guess.

Is there a way to access wheel RPM via the Python API?

kmeids, 2021-05-18:
I am also looking on how to get the wheel's speed as Hello HarishKumar051993.
Any help on this topic would be appreciated!

kmeids, 2021-05-18:
@HarishKumar051993 @corkyw10 just referencing the following related issue: [#2989 ]( https://github.com/carla-simulator/carla/issues/2989)

closed 2023-06-02 by stale[bot]