carla-simulator/carla #4199: a get_spawn_points for walkers

qhaas, 2021-05-12:
Per the [spawnnpc](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/examples/spawn_npc.py#L200) example, the suggested way to get spawn points for walkers is to use `get_random_location_from_navigation`.

For vehicles, if one wants to avoid spawn point collisions and maximize the number of vehicles in a simulation, one can use [`get_spawn_points` to return a list of vehicular spawn points](https://carla.readthedocs.io/en/latest/python_api/#getters_9).

It would be useful to have a similar method that returns a list of suggested spawn points for walkers.  This would also solve Issue #2136 since users could use `random.choice` with a fixed seed to select from a list of walker spawn points.

qhaas, 2021-09-10:
I did not see a way to get 'suggested' spawn points in CARLA 0.9.12 like one can do with vehicles.  Given we don't know if there will be a collision or other interference with spawning walkers until we try to actually spawn them, it hampers our ability to know how many walkers we will get when we batch up the spawn actor commands.

Created the script below that iterates through each town, displays how many suggested vehicle spawn points there are (for reference) and attempts to spawn thousands of walkers at random locations.  Given the disparity between attempted spawns and successful spawns, we need a better way to select walker spawn points.
```
town Town01, supports 255 vehicles, successfully spawned 8718 walkers of 16384 attempts
town Town02, supports 101 vehicles, successfully spawned 9492 walkers of 16384 attempts
town Town03, supports 265 vehicles, successfully spawned 11161 walkers of 16384 attempts
town Town04, supports 372 vehicles, successfully spawned 11407 walkers of 16384 attempts
town Town05, supports 302 vehicles, successfully spawned 11466 walkers of 16384 attempts
town Town06, supports 436 vehicles, successfully spawned 9917 walkers of 16384 attempts
town Town07, supports 116 vehicles, successfully spawned 13331 walkers of 16384 attempts
town Town10HD, supports 155 vehicles, successfully spawned 12697 walkers of 16384 attempts
```
[findWalkerCap.txt](https://github.com/carla-simulator/carla/files/7146705/findWalkerCap.txt)

ricardodeazambuja, 2022-03-11:
I was trying to get feasible spawn points for walkers, so I decided to try get_random_location_from_navigation. However, it seems to be sampling always from the same map (?!?). I saw that after I modified the code from @qhaas to visualise the points:
* [getSpawnPoint4Walker.py](https://github.com/ricardodeazambuja/carla-simulator-python/blob/804c05416ebf754ecd41eb18e71b21795eb8b65e/my_modified_files/getSpawnPoint4Walker.py)

keeees, 2022-10-21:
Any update on this issue?

qhaas, 2024-12-05:
AFAIK, the latest CARLA 0.9.15 release does not have this feature.  Now that I'm working with carla again, still would find it useful.