carla-simulator/carla #4200: Windows10 build Carla failed

newxf2015, 2021-05-13:
make osm2odr
mingw\include\zconf.h(475): fatal error C1083: 无法打开包括文件: “unistd.h”: No such file or directory 
make PythonAPI 
C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stdlib.h(1015): note: 参见“wcstombs”的声明
d:\emulation\carla\pythonapi\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: 无法打开包括文件: “OSM2ODR.h”: No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Professional\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

newxf2015, 2021-05-13:
I fixed

closed 2021-05-13 by newxf2015

llyyaa, 2021-06-01:
> I fixed

How? I'm facing the same problem

LucasFKobernic, 2021-06-18:
Hi @newxf2015, 
For some reasons i needed to rebuild CARLA and siche then im  having the same error as you.

I would aprreciate if you could tell me how do u fixed it.

llyyaa, 2021-06-18:
> Hi @newxf2015,
> For some reasons i needed to rebuild CARLA and siche then im having the same error as you.
> 
> I would aprreciate if you could tell me how do u fixed it.

If you have Anaconda binary directories in your "Path" environment variables, remove them temporarily, then restart and build again. I solved the problem of OSM2ODR.h in this way.