carla-simulator/carla #4202: Fixed the server FPS -- asynchronous mode

liuyandong1988, 2021-05-13:
I want to run the carla server by fixed FPS. I read the carla document [synchrony and timestep](https://carla.readthedocs.io/en/0.9.9/adv_synchrony_timestep/). I try the code 
```
        settings = self.world.get_settings()
        settings.synchronous_mode = False
        settings.fixed_delta_seconds = 1.0/10.0
        self.world.apply_settings(settings)
```
I want the carla server FPS: 10. But the result is ... in the picture. I think this is the default mode (Variable time-step) in CARLA.
[image]
I want to fix the server FPS...

Client API version     = 0.9.9.4 
Simulator API version  = 0.9.9.4-dirty 
Client FPS is 20: clock.tick_busy_loop(20.0)

DSantosO, 2021-05-26:
Hello @liuyandong1988,
With the settings.fixed_delta_seconds, you are fixing the time delta between frames. This is the 'simulation' time that happens between two different frames. But of course, the simulator is going to try to give you as many FPS as it can. 
If you want to fix the FPS, you will need to do it in synchronous mode and control this from the client.

liuyandong1988, 2021-05-27:
> Hello @liuyandong1988,
> With the settings.fixed_delta_seconds, you are fixing the time delta between frames. This is the 'simulation' time that happens between two different frames. But of course, [...]

Thanks! You mean I cannot fix the server fps by asynchronous mode and fixed_delta_seconds. But carla documentation provides the variable step and fixed step for asynchronous mode ...

closed 2023-06-02 by stale[bot]