carla-simulator/carla #4204: Error when ingesting a map from RoadRunner

AIasd, 2021-05-14:
Hi I have been following the tutorial [here](https://carla.readthedocs.io/en/0.9.11/tuto_A_add_map/) to export a map from RoadRunner to Carla. In particular, I tried "Map ingestion in a build from source".

The exported map files look like the following:
[image]

Next, I tried to run `make import ARGS="--package straightroad --no-carla-materials"` to do the ingestion. However, the following error popped up:
```
parse-options: unrecognized option '--package'
parse-options: unrecognized option '--no-carla-materials'
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c3-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: CARLA version 0.9.11-5-g4c8f4d5f.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
parse-options: unrecognized option '--package'
parse-options: unrecognized option '--no-carla-materials'
BuildCarlaUE4.sh: Using Unreal Engine at '/home/zhongzzy9/Documents/DISK_SDA/sda/UnrealEngine-4.24'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/mnt/sda/carla/Unreal/CarlaUE4'
bash "/mnt/sda/UnrealEngine-4.24/Engine/Build/BatchFiles/Linux/RunMono.sh" "/mnt/sda/UnrealEngine-4.24/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/mnt/sda/carla/Unreal/CarlaUE4/CarlaUE4.uproject" --package straightroad --no-carla-materials

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
/mnt/sda/UnrealEngine-4.24/Engine /mnt/sda/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/mnt/sda/UnrealEngine-4.24).
Creating makefile for straightroad (no existing makefile)
ERROR: Couldn't find target rules file for target 'straightroad' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
       Location: /mnt/sda/UnrealEngine-4.24/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
       Target rules found:
Makefile:290: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/mnt/sda/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:83: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 2
```
What is the issue here?

Also, as a side note, I succeeded in building carla from source following the tutorial [here](https://carla.readthedocs.io/en/0.9.11/build_linux/) in the same folder.

jbirtel, 2021-05-14:
Did you try to use "--package=straightroad" or "--packages=straightroad" (so with the **s** and the **=** in both cases) ? I also had a problem with the ingestion, but the '=' solved it for me the first time. This is a bit confusing in the docs.

However, I would also like to report an issue under this headline.
I user Blender and Blender-OSM to get a fbx file and a OSM file from a specific coordinates frame. I converted the osm with CARLA using the tutorial [here](https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/) and received a XODR. Then I wanted to import them by also using make import. After about 1 or 2 minutes I get the error 'XODR file doesn't exist':

> XODR file doesn't exist, ignoring crosswalks from openDRIVE
> Traceback (most recent call last):
>   File "/home/jonas-bsi/carla/Util/BuildTools/Import.py", line 385, in <module>
>     main()
>   File "/home/jonas-bsi/carla/Util/BuildTools/Import.py", line 382, in main
>     import_assets_from_json_list(json_list)
>   File "/home/jonas-bsi/carla/Util/BuildTools/Import.py", line 280, in import_assets_from_json_list
>     build_binary_for_navigation(package_name, dirname, maps)
>   File "/home/jonas-bsi/carla/Util/BuildTools/Import.py", line 362, in build_binary_for_navigation
>     os.remove(fbx_path_target)
> UnboundLocalError: local variable 'fbx_path_target' referenced before assignment
> Util/BuildTools/Linux.mk:13: recipe for target 'import' failed
> make: *** [import] Error 1

AIasd, 2021-05-14:
Hi @jbirtel , thanks a lot for your help! After using the suggested "--packages=straightroad", the problem is solved!

corkyw10, 2021-05-17:
Glad you got the problem solved. This command will be updated in the next release of the documentation.

closed 2021-05-17 by corkyw10