carla-simulator/carla #4208: Building Carla on M1 Mac Air Big Sur w/ Unreal Engine 4.25.4

christosachilleoudis, 2021-05-14:
Hi Everyone! I was wondering whether anyone has tried building CARLA on the M1 Mac Air with M1 Chip. I read through issue #150 and read through @analog-cbarber's solution but came to the conclusion that these solutions will not work since the M1 Mac requires Xcode 12. 

Here are my currents specs:
laptop: MacBook Air M1 Chip
OS: macOS Big Sur (11.3.1)
Xcode: downloaded from the App Store (Version 12.5)

I installed Unreal Engine 4.25.4 successfully via the following article:
https://medium.com/techiepedia/setting-up-unreal-engine-m1-macbook-for-c-game-development-eb40c12237d1

Any help or insight into this issue would be much appreciated! Btw I am a total noob so please let me know if there is any helpful information that I have failed to add. 

Thanks!

ZHAO-Zirui, 2021-05-22:
Hi there, I'm trying to make carla python API on my MBP M1. For now I'm still trying to pass `Setup.sh` on carla building.
After many failed attempts, I believe that ther version of `llvm clang clang++` is the most important thing.

We know that `clang` and `clang++` in BigSur is `12.0.5` and `llvm` which install by homebrew is  also `12.0.5`. But CARLA needs to use version 8 (`clang-8 clang++-8 llvm-8`). So, if we can change macOS clang and llvm version to 8. It might be possible for using CARLA on M1 Mac.

And there is another way to use CARLA. You can run CARLA server on an headless server through docker or just run the server on your PC. Install virutal machine in your M1 mac with aarch64 ubuntu system. Build the CARLA python API and use it. (You can also ask me if you need CARLA egg package for aarch64 linux) . 

I'm going to try something to change my Mac clang and llvm version. But I think it might be difficult. I hope we can share with each other if there is more progress

closed 2021-06-15 by christosachilleoudis

mariam-jamal, 2021-07-19:
Hi, I am trying to run CARLA on my Mac M1 air. But I don't know how to do it. The build instructions are only for linux and windows. I am a beginner so I am not able to follow the jargon, can you please help me figure out how to do it on my machine?

analog-cbarber, 2021-07-19:
If you are a beginner, I would not bother to try. It took me a week to get this working the last time I tried and it required some modifications to the code. Even if you were to get it to work, CARLA will still run better on a Linux/Windows machine with a decent gaming GPU.

gowtham07, 2022-05-06:
@ZHAO-Garry  were you able to change Mac clang and llvm version successfully for installation? I am new to CARLA and want to use that, which version should I download for mac m1 (the windows one or linux). Is it even possible to run CARLA on mac M1
Thanks!

gowtham07, 2022-05-06:
@analog-cbarber Hello, I want to use CARLA for a project. Can you please let me know the steps that needed to be followed for installation in mac m1 as there are no guidelines to be followed for m1 mac and there are different threads on github which are very confusing for me. Thanks in advance

CamdenSpehl, 2024-11-18:
@analog-cbarber maybe rather then saying "I did this but I am going to gatekeep it," you could provide some steps. I am not a beginner but can't figure it out and no, I do not have a linux/windows machine to use instead so its not an option

analog-cbarber, 2024-11-18:
I am not "gatekeeping" anything. I am just giving advice based on my experience. I cannot give you any simple (or even complex set of steps) to get it working for you. It took me a week of rather intense work, including debugging and fixing issues with the C++ code (I think it was mostly incorrect type declarations).  I have no idea what may have changed since the last time I got this to work. I have no idea what you might actually have to do to get it working this time.

nveshaan, 2025-07-10:
i found out a work around here => https://github.com/carla-simulator/carla/discussions/9037