carla-simulator/carla #4210: walker control problem

nuomizai, 2021-05-15:
After I spawned a walker, the walker's z axis value decreased automatically. I got the walker control by `actor.get_control()` and found the value was always:

`WalkerControl(direction=Vector3D(x=1.000000, y=0.000000, z=0.000000), speed=0.000000, jump=False)`

But the z-axis value for the walker was continuously decreasing, and the actor speed gotten by `actor.get_velocity()` decreased from `Vector3D(x=0.000000, y=0.000000, z=0.000000)` to `Vector3D(x=0.00000, y=0.000000, z=-40.000000)` and remained at  `Vector3D(x=0.00000, y=0.000000, z=-40.000000)`. Did anyone know where the problem was?

nuomizai, 2021-05-18:
I found the solution! It seemed that I set the `self._actor.set_simulate_physics(enabled=True)` by accident. If I changed the code as `self._actor.set_simulate_physics(enabled=False)`, then the problem would disappear! Did anyone know why CARLA should have this function `set_simulate_physics`?

corkyw10, 2021-05-25:
Hi @nuomizai, you can see one of the uses for `set_simulate_physics` in the [hybrid mode for Traffic Manager](https://carla.readthedocs.io/en/0.9.11/adv_traffic_manager/#hybrid-physics-mode), it can help with performance in simulations with a lot of vehicles.

closed 2021-08-05 by corkyw10