carla-simulator/carla #4213: CARLA 0.9.11 make PythonAPI and make launch problem

zyan003, 2021-05-16:
Hi all, I have some issues while I was trying to build CARLA. 
[image]

I have no idea to fix these issues. It is weird because rem is not found in batch file.

Thank you for your answer

corkyw10, 2021-05-25:
Hi @zyan003, can you share the documentation you are following?
Are you running the command in the Visual studio x64 native tools Command prompt?

zyan003, 2021-05-25:
Hi Corinne,

I solved this problem by resetting my Windows operating system. Then
everything works fine now. I did running the command in Visual Studio x64
native tools command prompt.

Best,
zyan003

corkyw10, 2021-05-25:
I'm glad you got it sorted out!

closed 2021-05-25 by corkyw10