carla-simulator/carla #4216: Import error when trying to import fbx and xodr from OSM and Blender

jbirtel, 2021-05-17:
Carla 0.9.11
Ubuntu 18.04.5 LTS

I am trying to make an import of an .fbx and .xodr file into Carla. Therefore, I copied coordinates of a frame I searched in OpenStreetMap and inserted them into Blender/Blender-OSM. Blender then automatically created a 3D model which I exported as fbx. Then, with the same coordinates, I created a .osm-file which I converted to an .xodr-file using Carla (using Step 2 of this tutorial [here](https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/)).
I then tried to import both in order to create a new Carla map by using this command:

`make import ARGS="--package=map_blender --no-carla-materials"
`

but I receive the following error after about 1-2 minutes:

...... many many of this LogModuleManager: Shutting down and abandoning module.
LogExit: Exiting.
XODR file doesn't exist, ignoring crosswalks from openDRIVE
Traceback (most recent call last):
  File "/carla/Util/BuildTools/Import.py", line 385, in <module>
    main()
  File "/carla/Util/BuildTools/Import.py", line 382, in main
    import_assets_from_json_list(json_list)
  File "/carla/Util/BuildTools/Import.py", line 280, in import_assets_from_json_list
    build_binary_for_navigation(package_name, dirname, maps)
  File "/carla/Util/BuildTools/Import.py", line 362, in build_binary_for_navigation
    os.remove(fbx_path_target)
UnboundLocalError: local variable 'fbx_path_target' referenced before assignment
Util/BuildTools/Linux.mk:13: recipe for target 'import' failed
make: *** [import] Error 1

I also tried to import by providing a json file with the following structure:
```
{
   "maps": [
      {
         "source": "./map_blender.fbx", 
         "use_carla_materials": false, 
         "name": "./map_blender", (I also tried without './')
         "xodr": "./map_blender.xodr"
      }
   ], 
   "props": []
}
```

Would be really cool if someone could have a look at this.

jbirtel, 2021-05-26:
I now also tried to import xodr and fbx files from RoadRunner, after I received a license for it. But also here, same mistake. Would really like to know, why it does not work. Even if I do not give any arguments to ```make import```, there is this error

werewolfdev, 2022-01-19:
@jbirtel are you using CARLA build from source

jbirtel, 2022-01-19:
@werewolfdev I was using CARLA build from source, I guess. Sorry it was some time ago and I do not remember completely. I know I figured out that the problem with RoadRunner was due to some export settings, that were not set correctly according to the needs of CARLA. So this worked for me in the end. But I did not get osm-> xodr and fbx working. If I was able to import them into CARLA, there was a little offset xodr and fbx and the streets from the fbx and the actual path for the vehicles to drive along are shifted somehow.

zhuyingxiang95, 2023-03-21:
hi Jbirtel, I am doing almost the same work as u have done before. I have successfully imported the fbx model which was exported from osm blender and the oxdr file into Carla. the only issue is that the coordinate of the fbx model doesn't match the opendrive map.  when i tried to do some further research , the " make import " command does't work any more.(nothing created in the folder "content "after "make import" ). that's quite sad.
[image]

closed 2023-06-02 by stale[bot]

florber, 2024-03-26:
> hi Jbirtel, I am doing almost the same work as u have done before. I have successfully imported the fbx model which was exported from osm blender and the oxdr file into Carla. the only issue is that [...]

I had the same offset, but was able to move and rotate the fbx content onto the oxdr content quiet easily.
As for not being able to run make import again: I had to rename the files in the import folder for that. Deleting the map and importing it again under the same name didnt seem to work.

Generally, I think that the blender OSM approach should be added to the wiki of Carla, its working much better than any other approaches i found, other than maybe the commerical roadrunner tool.