carla-simulator/carla #4221: Question regarding ingesting a UE4 modified map in a CARLA package

AIasd, 2021-05-17:
Hi,

From [the documentation of "add a new map"](https://carla.readthedocs.io/en/0.9.11/tuto_A_add_map/), I can successfully do Map ingestion in a CARLA package as well as Map ingestion in a build from source.

Now I am trying to add spawn points to a map generated from RoadRunner in UE4 and then ingest the modified map into a CARLA package. How should I achieve this?

In particular, I have the files grouped in a folder generated by RoadRunner as shown below.
 
[image]

Next, I open the files in UE4 editor via "Map ingestion in a build from source" from the [doc](https://carla.readthedocs.io/en/0.9.11/tuto_A_add_map/). After I add spawn points to the map, however, I am not sure how to make these changes to the initial RoadRunner generated files (those shown in the screenshot). 

Since I want to do "Map ingestion in a CARLA package" later, I guess I have to make changes to the RoadRunner generated files directly, right? (so I can use UE4 docker to generate the map package for import?) My confusion is that when I try to export the modified map it seems that I can only choose the format "fbx" and "obj". If say, I choose fbx and replace the original fbx, the change of added spawn points won't be reflected in the map of the generated package (since fbx does not contain spawn ponts information?). 

Am I following the correct procedures? Or am I missing something? I am quite new to this and will appreciate any help!

AIasd, 2021-05-25:
Hi @corkyw10 , thank you for your reply!
What value should I put for `<map_package>`? I keep encountering the following error message when I tried `map package`, `straight_road`, and `straightroad`: 

```
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/zhongzzy9/Documents/DISK_SDA/sda/UnrealEngine-4.24'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/mnt/sda/carla/Unreal/CarlaUE4'
bash "/mnt/sda/UnrealEngine-4.24/Engine/Build/BatchFiles/Linux/RunMono.sh" "/mnt/sda/UnrealEngine-4.24/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/mnt/sda/carla/Unreal/CarlaUE4/CarlaUE4.uproject" packages=straightroad

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
/mnt/sda/UnrealEngine-4.24/Engine /mnt/sda/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/mnt/sda/UnrealEngine-4.24).
Creating makefile for packages=straightroad (no existing makefile)
ERROR: Couldn't find target rules file for target 'packages=straightroad' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
       Location: /mnt/sda/UnrealEngine-4.24/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
       Target rules found:
Makefile:290: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/mnt/sda/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:83: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 2
```

When I import the map, I run `make import ARGS="--package straightroad --no-carla-materials"` and my directory structure for carla/Unreal/CarlaUE4/Content looks like:
```
├── Carla
├── Collections
├── Developers
└── map_package
    ├── Config
    │   └── map_package.Package.json
    ├── Maps
    │   └── straight_road
    │       ├── Asphalt1_Diff.uasset
    │       ├── Asphalt1_Norm.uasset
    │       ├── Asphalt1_Road.uasset
    │       ├── Asphalt1_Spec.uasset
    │       ├── LaneMarking1_Diff.uasset
    │       ├── LaneMarking1_Marking.uasset
    │       ├── LaneMarking1_Norm.uasset
    │       ├── LaneMarking1_Spec.uasset
    │       ├── LaneMarkingYellow1_Marking.uasset
    │       ├── OpenDrive
    │       │   └── straight_road.xodr
    │       ├── straight_road_BuiltData.uasset
    │       └── straight_road.umap
    └── Static
        └── Road
            └── straight_road
                ├── straight_road_Roads_MarkingNode.uasset
                └── straight_road_Roads_RoadNode.uasset
```

corkyw10, 2021-05-25:
Can you try just with `make package ARGS="packages=map_package"`? It looks like your package was imported with the default `map_package` name originally. You can see that here:
[image]

bernatx, 2021-05-25:
Just don't forget the double dash for the parameters, like:
`make package ARGS="--packages=map_package"`

AIasd, 2021-05-25:
Hi @corkyw10 and @bernatx , `make package ARGS="--packages=map_package"` works perfectly! Thanks a lot! 

I also think it might be great to consider to add this command to the end of "Add a new map-Map ingestion in a build from source" subsection in the documentation for other people's convenience.

closed 2021-05-25 by AIasd