carla-simulator/carla #4223: [carla 0.9.10] Collisition with ‘Unknown’

ChoByungSoo2, 2021-05-18:
operating system : window 10
carla version : 0.9.10
Unreal Engine : 4.24.3

There is no problem when load.fbx is uploaded, but the following problems occur when running the following mode for maps imported from objects fbx files such as trafficLight.fbx or building.fbx:  

PythonAPI : manual_control.py , no_rendering_mode.py , spawn_npc.py

Problem: 
1. Even if there is no object in the driving path, the car cannot drive forward as if it were hit by something. (It occurs in many parts of the map generated.) -> Show Collocation with 'Unknown'
2. Spawn point error occurs -> ex) If 300 vehicles are created, only 3 vehicles will be created.

This video is a video of a problem occurring.

[https://youtu.be/DA089Qfw3hk](url)

Note) I imported the fbx file with the following options

[image]
[image]

ChoByungSoo2, 2021-05-18:
Fixed the problem.

corkyw10, 2021-05-25:
Hi @ChoByungSoo2  since you fixed the problem, I will close this issue. You can reopen if needed.

closed 2021-05-25 by corkyw10