carla-simulator/carla #4225: Difficult to work on codebase due to build system issues

enricoschroeder, 2021-05-18:
Hi there,

I'm currently implementing a custom sensor model for CARLA and are hitting a lot of build errors when building in Visual Studio (command line build seems fine):

 ```cannot execute 'C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\c1xx.dll': command line is too long to fit in debug record```

This makes development rather unpleasant. According to a quick search, this is a Unreal issue and can be fixed by setting bLegacyPublicIncludePaths = false; in the project *.Build.cs (which is also the recommended standard setting). This however requires fixing include paths.

Also, there are a lot of warnings regarding hidden symbols which Unreal build system treats as errors by default (you disabled this behavior as well).

So my question is this: Are there any plans or ongoing work to overhaul the project setup? If not, I could provide a PR myself, but I wanted to contact you first and make sure that you would potentially merge my PR (I would not want to do the work in vain).

enricoschroeder, 2021-05-28:
Hi! So I went down the rabbit hole and fixed some of the issues... ;)

The error I posted above can indeed be fixed by setting bLegacyPublicIncludePaths = false and using full include paths everywhere (with this option enabled, UnrealBuildTool (UBT) appends all folders of the project to the include paths, leading to a command line for the compiler that is too long).

However, then the next, more severe issue started to show: Whenever I touched any code really, the build would fail at totally unrelated source files. I almost went crazy as I did not understand that at all. After some research, it turns out that there is an evil UBT feature called 'unity build' which is to blame. I've found this forums post [here](https://forums.unrealengine.com/t/how-to-compile-in-non-unity-mode/94863/5) that explains well what was going on. Short explanation: UBT by default combines several code files into a bigger source file to reduce compile times. The big issue with this is that it masks missing includes, undefined symbols and other sorts of errors in the sources (as problematic files may or may not by accident be combined into the same cpp file). If you then alter anything, the order in which files are combined is changed and the build may break randomly. Did none of you run into this issue before?

So I disabled unity build and it turns out that almost all source files contain multiple errors. Probably you always had this feature enabled so all these errors where masked by this source 'frankensteining'... Anyways, I fixed the breaking issues and it now compiles without unity build. As said, I could provide a PR but wanted before to check in and discuss if you are open to accepting such a PR (that touches a majority of the code base).

jimmyw404, 2022-08-15:
I ran into this issue. I believe the root cause is changing the source code files without doing a clean build. I was able to resolve this without a clean build by adding several includes to CarlaRecorder.cpp, including:

+#include "Components/BoxComponent.h"
+#include "Carla/Actor/ActorRegistry.h"
+#include "Carla/Game/CarlaEpisode.h"
+#include "Carla/Lights/CarlaLightSubsystem.h"
+#include "Carla/Vehicle/VehiclePhysicsControl.h"
+#include "Carla/Traffic/TrafficLightBase.h"
+#include "Carla/Traffic/TrafficLightGroup.h"
+#include "Carla/Vehicle/CarlaWheeledVehicle.h"