carla-simulator/carla #4227: Astar path between two waypoints

alice-st, 2021-05-18:
I am working with CARLA_0.9.11 and I am using ```grp.trace_route(origin, destination)```  to find the astar path between the two waypoints. The results I'm getting when I use the above mentioned function, are odd to say the least.

**Code Used:**

```python
client = carla.Client('localhost', 2000)
client.set_timeout(2.0)
world = client.get_world()
map = world.get_map()

origin = carla.Location(x=13.5, y=-13.7, z=0.0)
destination = carla.Location(x=180, y=240, z=0.5)

dao = GlobalRoutePlannerDAO(map, 2)
grp = GlobalRoutePlanner(dao)
grp.setup()

route = grp.trace_route(origin, destination)

waypoint_list = []
for waypoint in range(len(route)):
        waypoint_list.append(route[waypoint][0])

for waypoint in waypoint_list:
            world.debug.draw_string(waypoint.transform.location, 'O', draw_shadow=False, color=carla.Color(r=0, g=0, b=0),   life_time=300., persistent_lines=True)
```

**Results:**
[image]
[image]

raymanfx, 2021-05-19:
This is the roundabout in Town03, right? I can confirm this and similar issues. Depending on the route down-sampling rate, a path through the roundabout will contain waypoints on a different route (like shown in your second picture).

hishamyousry, 2021-07-12:
I am seeing similar behavior in both CARLA 0.9.9.4 and 0.9.11. In 0.9.9.4, this issue sometimes showed up even on straight roads, but in 0.9.11, I am only seeing this in the roundabout, where the route suddenly jumps to another lane.
[image]

anmigipi, 2021-07-21:
Can you please share the rest of the code, too?

hishamyousry, 2021-07-25:
[PythonAPI-overlay.zip](https://github.com/carla-simulator/carla/files/6874091/PythonAPI-overlay.zip)
I'm using the normal CARLA-0.9.11 PythonAPI library, but made some small modifications:
automatic_control.py: Set the initial spawn_point to be spawn_points[0] instead of some random spawn point, and set the destination to be spawn_point[5]
basic_agent.py: Added a small function to draw the waypoints like in the screenshot above.
After that, I just run the script using: python3 automatic_control.py --agent Basic
You'll find both modified files in the attached ZIP archive.

anmigipi, 2021-07-25:
Sorry to ask - but how does this implement A*?

closed 2021-11-18 by jackbart94