carla-simulator/carla #4228: OpenDRIVE files generated by OddLot

augri, 2021-05-19:
I have been trying to use the open source OpenDRIVE editor OddLot, which has received some attention in here. However, I could not successfully load simple maps generated by OddLot into Carla. Has anyone had this issue and know how to resolve it?

corkyw10, 2021-05-25:
Hi @augri, I haven't tried this myself. Could you share how you are trying to import, any documentation you are following and what errors you are experiencing?

jbirtel, 2021-05-26:
I would also be interested in the result of this issue. Also tried to use OddLOT for map creation, but it did not work. I spoke to developer of OddLOT, as Carla always exited with an error when I wanted to import a map, but he also could not help me (the tips he gave me did not resolve the issue)
I don't know if it is Carla or UE4 that somehow has a problem with the xodr created by OddLOT

Axel1092, 2021-05-26:
Hi @jbirtel, could you share an OpenDRIVE generated with that tool so that we can check what might be the issue?

jbirtel, 2021-05-26:
Sure. I needed to pack it into a zip, as .xodr files are not allowed somehow.
[oddlot_map_easy.zip](https://github.com/carla-simulator/carla/files/6547165/oddlot_map_easy.zip)

Axel1092, 2021-05-27:
I think I already see the problem. Road ids in this OpenDRIVE are strings instead of an interger. Our OpenDRIVE parser expects numbers here and not full names in the id fields as it makes queries much faster. Most software that generate OpenDRIVE data follow this integer id convention and put the full name in a specific field `name` meant for that.
For now, there is no support for the full name id convention in CARLA.

jbirtel, 2021-05-27:
Okay, thank you. OddLOT allows to export with different settings. I exported now with providing only numerical ids. However, this also did not work for me.
[oddlot_map_easy_numberid.zip](https://github.com/carla-simulator/carla/files/6553073/oddlot_map_easy_numberid.zip)

jbirtel, 2021-06-16:
Did you already find something on the new file, that I provided?

villimikko, 2021-06-16:
I would also be very itrested about this. Having same crashing issues.

Axel1092, 2021-06-16:
Hi @jbirtel, sorry for the late response. 
I am observing that the geometry in this OpenDRIVE is not defined for the total length of the road:
```         
<planView>
         <geometry length="19.99493329923575" y="291.420707432292" x="-540.9911455658666" hdg="6.263611950223343" s="1">
            <line/>
         </geometry>
</planView>
```
This is the definition of the geometry of one of the roads from your OpenDRIVE, it consist of a single line. The issue is the parameter `s="1"`. As a road can be defined by several geometries, the parameter `s` indicates the starting position of the geometry along the road length. In this case, the road is defined by a line from 1m from the beginning of the road. The problem is that there is a gap from s="0" to s="1" without geometry which causes the crash in our parser as we need to build the entire continuous geometry of the road.

jbirtel, 2021-06-16:
No problem :)
So, what do you think how could this be fixed? Do you think it would help that the s is set to 0 everywhere in the file? Or would it be necessary that the tool has a function to create continuous roads  ?

darld, 2021-06-21:
Maybe this can be create CARLA map. https://github.com/johschmitz/blender-driving-scenario-creator

closed 2023-06-02 by stale[bot]