carla-simulator/carla #4235: [GPS][CARLA] Wrong GPS conversions

ley-Hao, 2021-05-21:
I would like to report a problem in coordinates conversions when using the GPS/GNSS. While doing Cartesian -> angular -> Cartesian, we get a significant shift on the Y-axis.

While doing Cartesian -> Angular is CARLA's responsibility through [`carla.Map.transform_to_geolocation`](https://carla.readthedocs.io/en/0.9.11/python_api/#methods_27), we tried two different methods for angular -> Cartesian : with `robot_localization` (see [conversions file](https://github.com/cra-ros-pkg/robot_localization/blob/noetic-devel/include/robot_localization/navsat_conversions.h)) and a minimal example with `pymap3d` as suggested in this previous issue #2737

# Pure python minimal example

By @vtalpaert

```python
import unittest
import time

import numpy as np
import pymap3d as pm
import xml.etree.ElementTree as ET

import carla

class TestGps(unittest.TestCase):
    """Based on https://github.com/carla-simulator/carla/issues/2737
    """

    def setUp(self):
        self.client = carla.Client('localhost', 2000)
        self.client.set_timeout(2.0)
        self.world = self.client.get_world()
        self.carla_map = self.world.get_map()

    def change_town(self, town):
        print(town)
        self.client.load_world(town)
        time.sleep(2)
        self.world = self.client.get_world()
        self.carla_map = self.world.get_map()

    def get_gps_origin(self):
        opendrive_str = self.carla_map.to_opendrive()
        xml_tree = ET.fromstring(opendrive_str)
        lat_ref, lon_ref = None, None
        for geo_elem in xml_tree.find('header').find('geoReference').text.split(' '):
            if geo_elem.startswith('+lat_0'):
                lat_ref = float(geo_elem.split('=')[-1])
            elif geo_elem.startswith('+lon_0'):
                lon_ref = float(geo_elem.split('=')[-1])
        return lat_ref, lon_ref, 0.

    def test_gps_conversions_on_spawn_points(self):
        for town in ['Town01', 'Town02', 'Town03', 'Town04', 'Town05', 'Town06', 'Town07']:
            self.change_town(town)

            transforms = self.carla_map.get_spawn_points()
            geo_locations = [self.carla_map.transform_to_geolocation(
                spawn_tf.location) for spawn_tf in transforms]
            origin_latitude, origin_longitude, origin_altitude = self.get_gps_origin()
            print(origin_latitude, origin_longitude, origin_altitude)
            initial_coordinates = np.array(
                [(tf.location.x, tf.location.y, tf.location.z) for tf in transforms])
            converted_coordinates = np.array([pm.geodetic2enu(geo_loc.latitude, geo_loc.longitude, geo_loc.altitude, origin_latitude,
                                                              origin_longitude, origin_altitude, ell=pm.utils.Ellipsoid('wgs84')) for geo_loc in geo_locations])

            error = initial_coordinates - converted_coordinates
            print(np.mean(error, axis=0))
            print(np.std(error - np.mean(error, axis=0), axis=0))
```

```bash
Town01
0.0 0.0 0.0
[-9.24646097e-06  2.97596375e+02  6.91951211e-03]
[5.46863880e-06 2.19772448e+02 4.85565938e-03]
Town02
0.0 0.0 0.0
[-6.81042547e-06  4.17641156e+02  4.73719191e-03]
[5.46275418e-06 1.29641226e+02 2.30493103e-03]
Town03
0.0 0.0 0.0
[-1.34666621e-05 -5.26064856e+00  1.95923572e-03]
[4.30770311e-05 2.23983902e+02 1.39847287e-03]
Town04
0.0 0.0 0.0
[-5.78260098e-06 -1.72460864e+02  9.23116418e-03]
[4.15588248e-05 3.69469762e+02 7.04270094e-03]
Town05
0.0 0.0 0.0
[ 3.16266680e-06 -1.38316552e+01  1.67422368e-03]
[6.00473414e-06 1.90188177e+02 1.23744981e-03]
Town06
0.0 0.0 0.0
[-1.12223137e-05  1.73090866e+02  1.13922030e-02]
[1.20805326e-05 1.85252682e+02 9.51519946e-03]
Town07
0.0 0.0 0.0
[ 3.89275918e-06 -8.69032153e+01  1.67190173e-03]
[4.19012416e-06 1.87363485e+02 1.65588565e-03]
.
----------------------------------------------------------------------
Ran 1 test in 25.724s

OK
```

In the OpenDrive files, the reference is defined by `<geoReference><![CDATA[+proj=tmerc +lat_0=0 +lon_0=0 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs ]]></geoReference>`. The convention says WGS84, just like used by pymap3d.

How can we have an error average of 400m ?

# Issues with the ROS-bridge (ROS2 Eloquent)

Instead of the `pymap3d` library, I used the `navsat_transform` node from `robot_localization` ROS2 Eloquent package to do the angular -> Cartesian conversions

## Difference between GPS and real trajectories

[image]

[image]

these 2 figure shows the positions estimated given by gps and the real positions of our vehicle. arrows show the forward direction of our vehicle. As we can see, there are bias between red points or green points and blue points, which represent the real positions of vehicle and the positions estimated. red points are real vehicle positions from topic /tf, and green points are real vehicle positions from /odometry/wheel. these 2 type real positions are from different sources but they have exactly same value

## Fixed bias when immobile

[image]

As shown by this figure, the positions estimated by gps are always right behind the vehicle. the bias of gps is clearly shown here and this is what we want to fix.

```python
launch.actions.DeclareLaunchArgument(
    name='spawn_point_ego_vehicle',  # x y z roll pitch yaw
    default_value='133.0,-59,0.2,0,0,180'
),
```

Coordinate of vehicle spwaning in this test is [133,-59,0.2,0,0,180].

```json
{
      "type": "sensor.other.gnss",
      "id": "gnss1",
      "spawn_point": {"x": 1.0, "y": 0.0, "z": 2.0},
      "noise_alt_stddev": 0.05, "noise_lat_stddev": 0.00000025, "noise_lon_stddev": 0.00000025,  
      "noise_alt_bias": 0.0, "noise_lat_bias": 0.0, "noise_lon_bias": 0.0
},
```

this test is made with the configuration which is shown above. longitide noise and latitude noise are in degre and altitude noise is in meter. and there is clearly no noise bias from configuration file.

## Bias changes according to position on the map

Here is the error when following a trajectory 

[image]

[image]

the figure above shows the bias on the X axis and Y axis. and the figure below shows trajectory of vehicle. 

As we can see, represented by the first figure, the bias on the Y axis are more than that on axis X. but theoretically, there should be not bias on Y axis. and now, bias on Y axis is between 0.1 to 0.8 meters.

the yellow and green circle marked are sections which the vehicle moved on Y axis and X axis. in these sections, the bias on X axis or Y axis change obviously.

[image]

## Bias changes according to position on the map when immobile

I tested the bias with different spawn point of vehicle:

Spawn point: [100,100,0.2,0,0,180]
```python
launch.actions.DeclareLaunchArgument(
    name='spawn_point_ego_vehicle',  # x y z roll pitch yaw
    default_value='100.0,100,0.2,0,0,180'
),
```

[image]

Spawn point: [-100,100,0.2,0,0,180]
```python
launch.actions.DeclareLaunchArgument(
    name='spawn_point_ego_vehicle',  # x y z roll pitch yaw
    default_value='-100.0,100,0.2,0,0,180'
),
```
[image]

I only modified the spawn point of vehicle and nothing else. But results of gps are in different position of vehicle. these 2 tests clearly show that the vehicle's position influence the gps bias.

Thanks.

vtalpaert, 2021-05-21:
Worth noting that the issue is happening in CARLA 0.9.10 (because we use the ROS2 Eloquent version of the bridge)

Gilaine, 2021-07-04:
Hello @ley-Hao 

any news about this issue, is there any way to get rid of this bias?

VincentCheng24, 2022-05-02:
Carla uses the left-hand coordinates. while ENU uses the right-hand one. So the initial_coordinates need to change to 
`initial_coordinates = np.array([(tf.location.x, -tf.location.y, tf.location.z) for tf in transforms])`

Yet, there're still some error between the initial and the converted.

vtalpaert, 2022-05-02:
Hi @VincentCheng24 , could you give the output when considering the Euclidean distance as the error between the points, or at the least the absolute value instead of just the difference ?

VincentCheng24, 2022-05-04:
Hi @vtalpaert here is the distance error in meter
```
Town01
Mean distance error:  1.002320944448858
Distance error std:   0.7344406348971494

Town02
Mean distance error:  1.4026113319248008
Distance error std:   0.43539032317350096

Town03
Mean distance error:  0.5932342026932097
Distance error std:   0.46281857075395094

Town04
Mean distance error:  1.1596860857478732
Distance error std:   0.7271281313217032

Town05
Mean distance error:  0.5021692887738411
Distance error std:   0.3981655899729742

Town06
Mean distance error:  0.6547405536478685
Distance error std:   0.5440733676350943

Town07
Mean distance error:  0.5526080355482472
Distance error std:   0.4193697459775573

Town10HD
Mean distance error:  0.3671618162618558
Distance error std:   0.26824421252947606
```