carla-simulator/carla #4238: set_hybrid_physics_mode causes hero's velocity to be zero

qhaas, 2021-05-23:
Seemingly related to Issue #3683 .  The 'hero' vehicle's `carla.Vehicle::get_velocity()` will start returning zero while clearly having resumed driving after having stopped when [hybrid mode](https://carla.readthedocs.io/en/0.9.11/adv_traffic_manager/#hybrid-physics-mode) is enabled in CARLA 0.9.11.  The following example reproduces this behavior, it is based on the provided CARLA PythonAPI examples and can be seen after the first few frames.

```
#!/bin/python3

import carla
import traceback
import queue
import csv
import random
import math

try:
    random.seed(0)

    # setup client/server
    client = carla.Client('localhost', 2000)
    world = client.load_world('Town01')
    settings = world.get_settings()
    settings.fixed_delta_seconds = 1./20
    settings.synchronous_mode = True
    world.apply_settings(settings)
    world = client.reload_world(False)
    world.set_weather(getattr(carla.WeatherParameters, 'ClearNoon'))
    tm = client.get_trafficmanager()
    tm.set_synchronous_mode(True)
    tm.set_random_device_seed(0)
    tm.set_hybrid_physics_mode(True)
    
    # spawn actors
    actors = []
    bpLibrary = world.get_blueprint_library()
    # vehicle
    bp = bpLibrary.filter('vehicle.chevrolet.impala')[0]
    bp.set_attribute('role_name', 'hero')
    transform = world.get_map().get_spawn_points()[5]
    vehicle = world.spawn_actor(bp, transform)
    vehicle.set_autopilot(True)
    actors.append(vehicle)
    #camera
    bp = bpLibrary.find('sensor.camera.rgb')
    transform = carla.Transform(carla.Location(x=1.5, z=2.4))
    camera = world.spawn_actor(bp, transform, attach_to=vehicle)
    cameraQueue = queue.Queue()
    camera.listen(cameraQueue.put)
    actors.append(camera)

    print('simulation running')
    for _ in range(20*360):
        tick = world.tick()

        # save imagery
        dataFrame = cameraQueue.get(True, 1)
        if dataFrame.frame != tick:
            raise RuntimeError('world ({}) sensor ({}) frame number mismatch'.format(tick, dataFrame.frame))
        dataFrame.save_to_disk('_out/rgb_{:06d}.png'.format(tick))

        v = vehicle.get_velocity()
        currentSpeedKPH = 3.6*math.sqrt(v.x**2+v.y**2+v.z**2)
        with open('_out/telemetry.csv', 'a') as outFile:
            csvWriter = csv.writer(outFile)
            csvWriter.writerow([tick, vehicle.is_at_traffic_light(), currentSpeedKPH])
    print('simulation complete')

    print('cleaning up')
    camera.stop()
    client.apply_batch([carla.command.DestroyActor(x) for x in actors])
    settings.fixed_delta_seconds = None
    settings.synchronous_mode = False
    world.apply_settings(settings)
    tm.set_synchronous_mode(False)
    world.wait_for_tick()

except Exception as err:
    print(traceback.format_exc())
```

jackbart94, 2021-05-26:
Hi, this is probably happening because the hero is registered to the Traffic Manager. I'll push a fix as soon as possible, thanks!

qhaas, 2021-05-26:
Thanks for checking into this.  I only noticed it when looking at the logs (since it starts out correctly, possibly due to asio).  It appears to return velocity properly when one doesn't enable `set_hybrid_physics_mode`, even with TM (autopilot) enabled.

jackbart94, 2021-05-31:
I've fixed it, I'll try to have it merged inside of `dev` as soon as possible! Thanks for pointing this out again.

qhaas, 2021-08-16:
Looks to be fixed in 0.9.12, thanks @jackbart94 !

closed 2021-08-16 by qhaas