carla-simulator/carla #4239: Error by "make PythonAPI" and "make launch" on Windows

LucasFKobernic, 2021-05-24:
Hi everyone, 

I'm trying to install CARLA from source on my computer:

Windows 10
NVIDIA Quadro P2000
Python 3.7.4
GNU Make 4.3
Visual Studio Community 2017

I'm following the instruction from here: https://carla.readthedocs.io/en/latest/build_windows/

Until the installation for Unreal Engine all works fine. But i'm getting problem with **make PythonAPI** and consequently with **make launch**

After i run **make PythonAPI** I got at the end:

```
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\my_address\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
Python 3 not found!
Installed Pythons found by py Launcher for Windows
No Installed Pythons Found!

Requested Python version (3) not installed, use -0 for available pythons

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\my_address\carla\PythonAPI\carla\dist"!
```

**But I'm using Python3 and it is added to the PATH. Why did he not find Python?**

**And there is no directory called "dist" after a run make PythonAPI.** 

Then by running **make launch** i get a lot of erros C4458, C2665,  C2039, C3861 and at end:

`make: *** [Util/BuildTools/Windows.mk:20: CarlaUE4Editor] Error 6`

Can anyone help me with this?

Thanks,

corkyw10, 2021-05-25:
Hi @LucasFKobernic, the windows install will use the python launcher (py) to install the API. This utility is bundled with Python installation but not installed by default. Can you run the command:

```
py -3 --version
```
To see if it is installed correctly? If not, there's information [here](https://docs.python.org/3/using/windows.html#installation-steps) about installing it.

huatianxue1994, 2021-05-26:
> Hi @LucasFKobernic, the windows install will use the python launcher (py) to install the API. This utility is bundled with Python installation but not installed by default. Can you run the command: [...]

I install python3.7.4. But there is still a error when I input "make launch"
[image]
[image]
Could you help me to solve the problem? Thank you advanced!

LucasFKobernic, 2021-05-26:
Hi everyone, 

@corkyw10 thanks for the information. I cheked and you were right, i didnt have py launcher installed. I have installed it, an now make PythonAPI works! Thanks a lot!

But I'm still having problem with **make launch**:

```
>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  12
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "C:\Users\z239023\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Users\z239023\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\" --zlib-install-dir "C:\Users\my_username\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Users\z239023\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Users\z239023\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\Users\z239023\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\Users\z239023\carla\Build\recast-cdce4e-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\" --toolset msvc-14.1 --version 1.72.0 -j 12
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\Users\z239023\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\Users\z239023\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in C:\Users\z239023\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA 0.9.11:
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20201222_232b876.tar.gz

   Unzip it in the "C:\Users\z239023\carla\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in C:\Users\z239023\carla\Util\ContentVersions.txt.

[... 223 lines cut ...]
  C:/Users/my_username/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp(382): error C2039: "GameDir": Ist kein Element von "FPaths"
  D:\UnrealEngine-release\Engine\Source\Runtime\Core\Public\Misc/Paths.h(12): note: Siehe Deklaration von "FPaths"
  C:/Users/my_username/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp(382): error C3861: "GameDir": Bezeichner wurde nicht gefunden.
  d:\unrealengine-release\engine\source\runtime\experimental\chaos\public\Chaos/PBDRigidsSOAs.h(30): error C2039: "_InterlockedCompareExchange": Ist kein Element von "FWindowsPlatformAtomics"
  D:\UnrealEngine-release\Engine\Source\Runtime\Core\Public\Windows/WindowsPlatformAtomics.h(13): note: Siehe Deklaration von "FWindowsPlatformAtomics"
  d:\unrealengine-release\engine\source\runtime\experimental\chaos\public\Chaos/PBDRigidsSOAs.h(30): error C2665: "_InterlockedCompareExchange": Durch keine der 4 Überladungen konnten alle Argumenttypen konvertiert werden.
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um\winbase.h(9450): note: kann "unsigned __int64 _InterlockedCompareExchange(volatile unsigned __int64 *,unsigned __int64,unsigned __int64)" sein
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um\winbase.h(9437): note: oder "unsigned long _InterlockedCompareExchange(volatile unsigned long *,unsigned long,unsigned long)"
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um\winbase.h(9426): note: oder "unsigned int _InterlockedCompareExchange(volatile unsigned int *,unsigned int,unsigned int)"
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\INCLUDE\intrin0.h(141): note: oder "long _InterlockedCompareExchange(volatile long *,long,long)"
[... cut ...]
  d:\unrealengine-release\engine\source\runtime\experimental\chaos\public\Chaos/PBDRigidsSOAs.h(36): note: bei Anpassung der Argumentliste "(volatile int8 *, int, int)"
  C:/Users/my_username/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveGenerator.cpp(19): fatal error C1083: Datei (Include) kann nicht geöffnet werden: "PhysicsEngine/BodySetupEnums.h": No such file or directory
make: *** [CarlaUE4Editor] Fehler 6
```
I'm getting this error 6. I have the Unreal Engine installed in another disk. Is that maybe the problem?

Thanks,

corkyw10, 2021-05-26:
Hey @LucasFKobernic, I just noticed you are using Make 4.3, however our documentation specifies that you should use 3.81, otherwise errors may occur. I am not sure though that this is what is causing the issue right now. 

i have a similar issue a few times recently because people have been confusing the installations between the most recent release (0.9.11) and the dev branch version of carla (which is what the documentation you linked points to). The 0.9.11 version requires our modified version of UE4.24 and the dev version requires our modified version of UE4.26. Can I just check that you haven't gotten the two mixed up?

LucasFKobernic, 2021-05-26:
Hi @corkyw10,

Tahnks for the answer.

About the make version,  i had already downgraded it to version 3.81. I had installed the modified version of UE4.26. So I uninstalled CARLA version 0.9.11 and installed the latest one. **make Python** works well, but i still getting error by **make launch**:

```
Building 27 actions with 12 processes...
  [1/27] SharedPCH.UnrealEd.cpp
  [2/27] Module.Carla.gen.1_of_6.cpp
  [3/27] Module.Carla.gen.4_of_6.cpp
  [4/27] Module.Carla.gen.5_of_6.cpp
  [5/27] PCH.CarlaUE4.cpp
  [6/27] CarlaUE4.cpp
  [7/27] UE4Editor-CarlaUE4.lib
     Bibliothek "C:\my_username\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\UE4Editor\Development\CarlaUE4\UE4Editor-CarlaUE4.lib" und Objekt "C:\my_username\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\UE4Editor\Development\CarlaUE4\UE4Editor-CarlaUE4.exp" werden erstellt.
  [8/27] Module.Carla.gen.3_of_6.cpp
  [9/27] Module.Carla.gen.2_of_6.cpp
  [10/27] Module.CarlaExporter.cpp
  C:/my_username/carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(153): warning C4996: 'AActor::GetComponentsByClass': Use one of the GetComponents implementations as appropriate Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\UnrealEngine-carla\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(3056): note: Siehe Deklaration von "AActor::GetComponentsByClass"
  [11/27] UE4Editor-CarlaExporter.lib
     Bibliothek "C:\my_username\carla\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.lib" und Objekt "C:\my_username\carla\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.exp" werden erstellt.
  [12/27] UE4Editor-CarlaExporter.dll
     Bibliothek "C:\my_username\carla\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.suppressed.lib" und Objekt "C:\my_username\carla\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.suppressed.exp" werden erstellt.
  [13/27] Module.Carla.cpp
  C:/my_username/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp(58): warning C4458: Deklaration von "TerrainNodeMaterial" blendet Klassenmember aus
  c:\users\z239023\carla\unreal\carlaue4\plugins\carla\source\carla\commandlet\PrepareAssetsForCookingCommandlet.h(161): note: Siehe Deklaration von "UPrepareAssetsForCookingCommandlet::TerrainNodeMaterial"
  C:/my_username/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp(382): error C2039: "GameDir": Ist kein Element von "FPaths"
  D:\UnrealEngine-carla\Engine\Source\Runtime\Core\Public\Misc/Paths.h(12): note: Siehe Deklaration von "FPaths"
  C:/my_username/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp(382): error C3861: "GameDir": Bezeichner wurde nicht gefunden.
  d:\unrealengine-carla\engine\source\runtime\experimental\chaos\public\Chaos/PBDRigidsSOAs.h(30): error C2039: "_InterlockedCompareExchange": Ist kein Element von "FWindowsPlatformAtomics"
  D:\UnrealEngine-carla\Engine\Source\Runtime\Core\Public\Windows/WindowsPlatformAtomics.h(13): note: Siehe Deklaration von "FWindowsPlatformAtomics"
  d:\unrealengine-carla\engine\source\runtime\experimental\chaos\public\Chaos/PBDRigidsSOAs.h(30): error C2665: "_InterlockedCompareExchange": Durch keine der 4 Überladungen konnten alle Argumenttypen konvertiert werden.
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um\winbase.h(9450): note: kann "unsigned __int64 _InterlockedCompareExchange(volatile unsigned __int64 *,unsigned __int64,unsigned __int64)" sein
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um\winbase.h(9437): note: oder "unsigned long _InterlockedCompareExchange(volatile unsigned long *,unsigned long,unsigned long)"
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um\winbase.h(9426): note: oder "unsigned int _InterlockedCompareExchange(volatile unsigned int *,unsigned int,unsigned int)"
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\INCLUDE\intrin0.h(141): note: oder "long _InterlockedCompareExchange(volatile long *,long,long)"
  d:\unrealengine-carla\engine\source\runtime\experimental\chaos\public\Chaos/PBDRigidsSOAs.h(30): note: bei Anpassung der Argumentliste "(volatile int8 *, int, int)"
  d:\unrealengine-carla\engine\source\runtime\experimental\chaos\public\Chaos/PBDRigidsSOAs.h(36): error C2039: "_InterlockedCompareExchange": Ist kein Element von "FWindowsPlatformAtomics"
  D:\UnrealEngine-carla\Engine\Source\Runtime\Core\Public\Windows/WindowsPlatformAtomics.h(13): note: Siehe Deklaration von "FWindowsPlatformAtomics"
  d:\unrealengine-carla\engine\source\runtime\experimental\chaos\public\Chaos/PBDRigidsSOAs.h(36): error C2665: "_InterlockedCompareExchange": Durch keine der 4 Überladungen konnten alle Argumenttypen konvertiert werden.
  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um\winbase.h(9450): note: kann "unsigned __int64 _InterlockedCompareExchange(volatile unsigned __int64 *,unsigned __int64,unsigned __int64)" sein
[... cut ...]
     Bibliothek "D:\UnrealEngine-carla\Engine\Plugins\Online\OnlineSubsystemUtils\Intermediate\Build\Win64\UE4Editor\Development\OnlineSubsystemUtils\UE4Editor-OnlineSubsystemUtils.suppressed.lib" und Objekt "D:\UnrealEngine-carla\Engine\Plugins\Online\OnlineSubsystemUtils\Intermediate\Build\Win64\UE4Editor\Development\OnlineSubsystemUtils\UE4Editor-OnlineSubsystemUtils.suppressed.exp" werden erstellt.
make: *** [CarlaUE4Editor] Fehler 6
```
Its still shwoing me error at action 13m but this time it goes until action 23 before crashing. Do you have any idea, what the problem might be?

Thanks for the help,

bernatx, 2021-05-26:
If you use the latest development source, then you need to use our fork of the Unreal Engine 4.26, and compile the engine.

bernatx, 2021-05-26:
You downloaded the repo as a .zip, but did you checkout the **carla** branch before compiling it?

LucasFKobernic, 2021-05-28:
Hi @bernatx 

I uninstalled and installed CARLA and Unreal again, ensuring the versions were correct and now it works! Thank you very much for your help.

closed 2021-05-28 by LucasFKobernic