carla-simulator/carla #4241: Engine RPM in Carla

HarishKumar051993, 2021-05-24:
How to Calculate the Engine RPM of a Carla Vehicle?

There is no API available to calculate Engine RPM.

jackbart94, 2021-05-28:
If you look into carla.VehiclePhysicsControl and check max_rpm (float), it gives out the maximum RPM of the vehicle's engine. Then, if you apply a throttle of 0.7 you are actually setting an RPM of 70% with respect to the maximum.

closed 2021-11-18 by jackbart94

YuhengZhi, 2024-08-28:
> If you look into carla.VehiclePhysicsControl and check max_rpm (float), it gives out the maximum RPM of the vehicle's engine. Then, if you apply a throttle of 0.7 you are actually setting an RPM of 70% with respect to the maximum.

In PhysX_3.4, throttle scales the maximum torque given by torque_curve(current_RPM), not directly RPM itself.