carla-simulator/carla #4242: 0.9.10, make launch failure: The system cannot find the path specified. Already defined the system variable

Ryan-Doren, 2021-05-25:
Hi, everyone,
I was trying to build Carla 0.9.11 on my Windows 10 system according to [the document v:  0.9.11](https://carla.readthedocs.io/en/0.9.11/build_windows/#necessary-software).  I have installed git, cmake, Visual Studio 2017, python3.7.8, and UE 4.24.3, downloaded and unzipped the assets, and succeeded when running `make PythonAPI`. I've also defined the environment variable UE4_ROOT as C:\Program Files\Epic Games\UE4.24.

However, when I run `make launch`, the following error occurred:

> -[BuildCarlaUE4]: [Batch params]: --build --all
> --build
> --all
> ECHO 处于关闭状态。
> -[BuildCarlaUE4]: Building Unreal Editor...
> 系统找不到指定的路径。
> make: *** [CarlaUE4Editor] 错误 1

The full output is as follows：

> A:\Carla>make launch
> -[Setup]: [Batch params]: --boost-toolset msvc-14.1
> -[Setup]: Asynchronous jobs:  16
> -[Setup]: Boost toolset:      msvc-14.1
> -[Setup]: Install directory:  "A:\Carla\Build\"
> -[Setup]: Installing zlib...
>     -[install_zlib]: [Batch params]: --build-dir "A:\Carla\Build\"
>     -[install_zlib]: A zlib installation already exists.
>     -[install_zlib]: Delete "A:\Carla\Build\zlib-install\" if you want to force a rebuild.
>     -[install_zlib]: Exiting...
> -[Setup]: Installing libpng...
>     -[install_libpng]: [Batch params]: --build-dir "A:\Carla\Build\" --zlib-install-dir "A:\Carla\Build\zlib-install\"
>     -[install_libpng]: A libpng installation already exists.
>     -[install_libpng]: Delete "A:\Carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
>     -[install_libpng]: Exiting...
> -[Setup]: Installing rpclib...
>     -[install_rpclib]: [Batch params]: --build-dir "A:\Carla\Build\"
>     -[install_rpclib]: A rpclib installation already exists.
>     -[install_rpclib]: Delete "A:\Carla\Build\rpclib-install\" if you want to force a rebuild.
>     -[install_rpclib]: Exiting...
> -[Setup]: Installing Google Test...
>     -[install_gtest]: [Batch params]: --build-dir "A:\Carla\Build\"
>     -[install_gtest]: A Google Test installation already exists.
>     -[install_gtest]: Delete "A:\Carla\Build\gtest-install\" if you want to force a rebuild.
>     -[install_gtest]: Exiting...
> -[Setup]: Installing "Recast & Detour"...
>     -[install_recast]: [Batch params]: --build-dir "A:\Carla\Build\"
>     -[install_recast]: A "Recast & Detour" installation already exists.
>     -[install_recast]: Delete "A:\Carla\Build\recast-cdce4e-install\" if you want to force a rebuild.
>     -[install_recast]: Exiting...
> -[Setup]: Installing Boost...
>     -[install_boost]: [Batch params]: --build-dir "A:\Carla\Build\" --toolset msvc-14.1 --version 1.72.0 -j 16
>     -[install_boost]: A Boost installation already exists.
>     -[install_boost]: Delete "A:\Carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
>     -[install_boost]: Exiting...
> 已复制         1 个文件。
> 已复制         1 个文件。
> -[Setup]: Installing Xercesc...
>     -[install_xercesc]: [Batch params]: --build-dir "A:\Carla\Build\"
>     -[install_xercesc]: A xerces installation already exists.
>     -[install_xercesc]: Delete "A:\Carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
>     -[install_xercesc]: Exiting...
> fatal: not a git repository (or any of the parent directories): .git
> -[Setup]: Creating "CMakeLists.txt.in"...
> -[Setup]:
>    ###########
>    # SUCCESS #
>    ###########
> 
>    IMPORTANT!
> 
>    All the CARLA library dependences should be installed now.
>    (You can remove all "*-src" folders in A:\Carla\Build\ directory)
> 
>    You only need the ASSET PACK with all the meshes and textures.
[... 226 lines cut ...]
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(256): error C2143: 语法错误: 缺少“;”(在“=”的前面)
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(256): error C2143: 语法错误: 缺少“)”(在“=”的前面)
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(256): error C2059: 语法错误:“=”
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(256): error C2065: “i”: 未声明的标识符
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(256): error C2065: “mesh”: 未声明的标识符
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(256): error C2059: 语法错误:“)”
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(259): error C2065: “indexBuffer”: 未声明的标识符
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(259): error C2059: 语法错误:“;”
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(262): error C2065: “nbVerts”: 未声明的标识符
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(153): warning C4996: 'AActor::GetComponentsByClass': Use one of the GetComponents implementations as appropriate Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
> 1>  C:\Program Files\Epic Games\UE4.24\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(3056): note: 参见“AActor::GetComponentsByClass”的声明
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(266): error C2059: 语法错误:“}”
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(266): error C2143: 语法错误: 缺少“;”(在“}”的前面)
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(267): error C2143: 语法错误: 缺少“;”(在“}”的前面)
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(267): error C2059: 语法错误:“}”
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(269): error C2059: 语法错误:“}”
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(269): error C2143: 语法错误: 缺少“;”(在“}”的前面)
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(272): error C2143: 语法错误: 缺少“;”(在“{”的前面)
> 1>A:/Carla/Unreal/CarlaUE4/Plugins/CarlaExporter/Source/CarlaExporter/Private/CarlaExporter.cpp(272): error C2447: “{”: 缺少函数标题(是否是老式的形式表?)
> 1>A:\VisualStudio\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3073: 命令“"C:\Program Files\Epic Games\UE4.24\Engine\Build\BatchFiles\Build.bat" -Target="CarlaUE4Editor Win64 Development -Project=\"A:\Carla\Unreal\CarlaUE4\CarlaUE4.uproject\"" -Target="ShaderCompileWorker Win64 Development -Quiet" -WaitMutex -FromMsBuild”已退出，代码为 6。
> 1>已完成生成项目“CarlaUE4.vcxproj”的操作 - 失败。
> ========== 生成: 成功 0 个，失败 1 个，最新 2 个，跳过 0 个 ==========

I have no idea how to fix it. I would be grateful if you can help me with it.
@bernatx @marcgpuig @Axel1092 @nsubiron

winterwindwang, 2021-07-12:
Any solution here? I have encountered this question as well.

Ryan-Doren, 2021-07-12:
> Any solution here? I have encountered this question as well.

Sorry to tell you that I’ve switched to Ubuntu 18.04 and after a little bit  debugging, I succeeded in making launch on Linux. It seems that you’ll never know what kind of problems is going to happen to Windows operation system.😓

winterwindwang, 2021-07-12:
Thank you for your quickly response. I solve this problem with: "Go to: HKEY_LOCAL_MACHINE\SOFTWARE\EpicGames\Unreal Engine%UE_VERSION% (replace %UE_VERSION% with 4.22) and add a new String value with Value name InstalledDirectory and Value data your path to UE_4.24 (for me it is E:\games\epic\UnrealEngine\)", after I go through this(https://github.com/carla-simulator/carla/issues/1941#issuecomment-625375872) soultion.

Ryan-Doren, 2021-07-12:
> Thank you for your quickly response. I solve this problem with: "Go to: HKEY_LOCAL_MACHINE\SOFTWARE\EpicGames\Unreal Engine%UE_VERSION% (replace %UE_VERSION% with 4.22) and add a new String value with Value name InstalledDirectory and Value data your path to UE_4.24 (for me it is E:\games\epic\UnrealEngine\)", after I go through this([#1941 (comment)](https://github.com/carla-simulator/carla/issues/1941#issuecomment-625375872)) soultion.
> […](#)

Thank you very much for your solution!

closed 2022-04-29 by stale[bot]