carla-simulator/carla #4259: Cannot Run Automatic_control.py in Carla

Michaelll268, 2021-05-28:
I tried everything with the pythonpath and stuffs but I still cant load automatic_control.py in carla,, the error that is mentioned is "No Module named agents.navigation". Please Help me,, thanks a lot

jackbart94, 2021-05-28:
We need more information. Did you do ``make PythonAPI`` and ``make LibCarla`` ? Did you check that the path in the ``automatic_control.py`` script correctly points to your CARLA folder structure? Maybe you've made some changes there.

Michaelll268, 2021-05-28:
@jackbart94  how to make pythonAPI and make LibCarla, is it directly on the CMD or in the corresponding folder of the file?

jackbart94, 2021-05-31:
Your folder structure is probably wrong. Check the code in ``automatic_control.py`` and check out where it is trying to find the agents folder and why it is failing. Which version of CARLA are you using?

closed 2021-11-18 by jackbart94