carla-simulator/carla #4261: Is it possible to Hide ego vehicle during data acquisition from camera in synchronous mode?

chowkamlee81, 2021-05-29:
I want to hide ego vehicle for data acquisition from camera in synchronous mode. Kindly help

DSantosO, 2021-06-01:
Hello @chowkamlee81,

Right now it is not possible to do that from the Python API, but if you are willing to modify the source of the simulator it should not be that difficult to add a method to control that. We can give you some hints about that.
Also, another easier solution, if you are in the editor you can set the Visible field to 'false' to any vehicle of CARLA or one that you have imported so whenever you spawned everything will work the same but the car won't be rendered.

lucalazzaroni, 2022-11-30:
Hi @DSantosO , would it be possible in a simulation to hide the ego vehicle only frome the lidar sensor? But keeping for examples collisions on.
Thank you in advance for your help

closed 2023-06-02 by stale[bot]