carla-simulator/carla #4262: Loading Town10HD crashes program - Vulkan Issue?

sshashank124, 2021-05-30:
Carla Version: dev branch
OS: Ubuntu 18.04
GPU: RTX 2060 Super
---

Trying to load level "Town10HD" causes the program to crash with the following error. It seems to be an issue with the Vulkan backend:

```
Fatal error: [File:/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanUtil.cpp] [Line: 803] VulkanRHI::vkDeviceWaitIdle(Device) failed, VkResult=-4 at /home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:1173 with error VK_ERROR_DEVICE_LOST 0x00007fcd24c2badf libUE4Editor-VulkanRHI.so!VulkanRHI::VerifyVulkanResult(VkResult, char const*, char const*, unsigned int) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanUtil.cpp:802] 0x00007fcd24b9eae3 libUE4Editor-VulkanRHI.so!FVulkanDevice::WaitUntilIdle() [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:1173] 0x00007fcd24c54dc6 libUE4Editor-VulkanRHI.so!FVulkanViewport::DestroySwapchain(FVulkanSwapChainRecreateInfo*) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanViewport.cpp:711] 0x00007fcd24c565dc libUE4Editor-VulkanRHI.so!FVulkanViewport::RecreateSwapchainFromRT(EPixelFormat) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanViewport.cpp:613] 0x00007fcd24c5db90 libUE4Editor-VulkanRHI.so!TEnqueueUniqueRenderCommandType<FVulkanDynamicRHI::RHIResizeViewport(FRHIViewport*, unsigned int, unsigned int, bool, EPixelFormat)::ResizeViewportName, FVulkanDynamicRHI::RHIResizeViewport(FRHIViewport*, unsigned int, unsigned int, bool, EPixelFormat)::$_191>::DoTask(ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/RenderCore/Public/RenderingThread.h:183] 0x00007fcd24c5deec libUE4Editor-VulkanRHI.so!TGraphTask<TEnqueueUniqueRenderCommandType<FVulkanDynamicRHI::RHIResizeViewport(FRHIViewport*, unsigned int, unsigned int, bool, EPixelFormat)::ResizeViewportName, FVulkanDynamicRHI::RHIResizeViewport(FRHIViewport*, unsigned int, unsigned int, bool, EPixelFormat)::$_191> >::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Async/TaskGraphInterfaces.h:886] 0x00007fce30dbbcac libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksNamedThread(int, bool) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:709] 0x00007fce30dba3ae libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksUntilQuit(int) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:600] 0x00007fce2b5e355f libUE4Editor-RenderCore.so!RenderingThreadMain(FEvent*) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp:372] 0x00007fce2b62e502 libUE4Editor-RenderCore.so!FRenderingThread::Run() [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp:526] 0x00007fce30e6be17 libUE4Editor-Core.so!FRunnableThreadPThread::Run() [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25] 0x00007fce30e301e3 libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:185] 0x00007fce317ba6db libpthread.so.0!UnknownFunction(0x76da) 0x00007fce279e771f libc.so.6!clone(+0x3e)

libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
libUE4Editor-VulkanRHI.so!VulkanRHI::VerifyVulkanResult(VkResult, char const*, char const*, unsigned int) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanUtil.cpp:802]
libUE4Editor-VulkanRHI.so!FVulkanDevice::WaitUntilIdle() [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:1173]
libUE4Editor-VulkanRHI.so!FVulkanViewport::DestroySwapchain(FVulkanSwapChainRecreateInfo*) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanViewport.cpp:711]
libUE4Editor-VulkanRHI.so!FVulkanViewport::RecreateSwapchainFromRT(EPixelFormat) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanViewport.cpp:613]
libUE4Editor-VulkanRHI.so!TEnqueueUniqueRenderCommandType<FVulkanDynamicRHI::RHIResizeViewport(FRHIViewport*, unsigned int, unsigned int, bool, EPixelFormat)::ResizeViewportName, FVulkanDynamicRHI::RHIResizeViewport(FRHIViewport*, unsigned int, unsigned int, bool, EPixelFormat)::$_191>::DoTask(ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/RenderCore/Public/RenderingThread.h:183]
libUE4Editor-VulkanRHI.so!TGraphTask<TEnqueueUniqueRenderCommandType<FVulkanDynamicRHI::RHIResizeViewport(FRHIViewport*, unsigned int, unsigned int, bool, EPixelFormat)::ResizeViewportName, FVulkanDynamicRHI::RHIResizeViewport(FRHIViewport*, unsigned int, unsigned int, bool, EPixelFormat)::$_191> >::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Async/TaskGraphInterfaces.h:886]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksNamedThread(int, bool) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:709]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksUntilQuit(int) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:600]
libUE4Editor-RenderCore.so!RenderingThreadMain(FEvent*) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp:372]
libUE4Editor-RenderCore.so!FRenderingThread::Run() [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp:526]
libUE4Editor-Core.so!FRunnableThreadPThread::Run() [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/phaqlow/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:185]
libpthread.so.0!UnknownFunction(0x76da)
libc.so.6!clone(+0x3e)
```

Any ideas on how to resolve this? Could this be an issue of insufficient memory?

sshashank124, 2021-06-07:
It was suggested that the latest changes to dev branch might have fixed this. However, I still get the same crash on the dev branch as of today

PaulHSK, 2021-11-15:
same bug here.
working on a fresh 18.04 lts with RTX 3000

got same error on other Hardware (System is on a external SSD)
tried different drivers (495 & 470) and turned off internal grafic card

Do you get this Error Message on "Make Launch" as-well?

```
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: === Handled ensure: ===
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: 
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/paul/UnrealEngine-carla/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: 
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: Stack: 
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x00007f857f114e83 libUE4Editor-VulkanRHI.so!FVulkanDevice::FVulkanDevice(FVulkanDynamicRHI*, VkPhysicalDevice_T*) [/home/paul/UnrealEngine-carla/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:182]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x00007f857f1ae8a6 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::SelectAndInitDevice() [/home/paul/UnrealEngine-carla/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:551]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x00007f857f1aabd3 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::InitInstance() [/home/paul/UnrealEngine-carla/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:694]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x00007f857f1aa7e8 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::Init() [/home/paul/UnrealEngine-carla/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:241]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x00007f865190300d libUE4Editor-RHI.so!RHIInit(bool) [/home/paul/UnrealEngine-carla/Engine/Source/Runtime/RHI/Private/DynamicRHI.cpp:228]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x0000000000283ba3 UE4Editor!FEngineLoop::PreInitPreStartupScreen(char16_t const*) [/home/paul/UnrealEngine-carla/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:2555]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x000000000027aacc UE4Editor!GuardedMain(char16_t const*) [/home/paul/UnrealEngine-carla/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x00007f864e3b8286 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/paul/UnrealEngine-carla/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x00007f864e1de0b3 libc.so.6!__libc_start_main(+0xf2)
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: [Callstack] 0x0000000000263d39 UE4Editor!_start()
[2021.11.15-13.36.05:278][  0]LogOutputDevice: Error: 

```

weijiepengx, 2021-12-23:
I met this issue while I launch with `--gdb` .
working on Ubuntu20.04lts with 3090.

Hardliu, 2022-01-02:
I have ubuntu20.04&gtx3080ti&495.25&carla-09.12 with same issue.

Hardliu, 2022-01-02:
> Do you get this Error Message on "Make Launch" as-well?

Do you search method to sovle it? I have same issue with you.

closed 2023-06-02 by stale[bot]

evoevo07, 2024-04-08:
Same issue here...