carla-simulator/carla #4265: Multiple realtime speed and Physicsengine/VehicleController

semmelknoedel, 2021-06-01:
Hi,

I was wondering if Unreal physics scale with the speed when the simulation runs faster than realtime.

My setup:
I use Carla 0.9.11 and a modified version of the scenario runner.
The modification is mainly, so I can run a scenario without running a second script for the ego vehicle, since I can view whats happening in the Carla window and have recorded data. For that I added a new Behavior for a custom agent based on the BasicAgent.
Since this was running without pygame and with a frame rate of 20 fps (in synchronous mode) while my computer is capable of 200 fps, the simulation was running pretty fast (which wouldn't bother me since the timestamps in the recording verify that there are only 20 fps).
However, the VehicleController seems not to scale with this speed. If I give parameters for a PID-Controller at a frame rate of 20 fps and it is working fine, I assumed that the speed should not matter, but instead for the controller the parameters have to be adjusted to 200 fps. I had to include a pygame-wait to slow down the simulation speed to realtime of 20fps in order to make it work.

Is this enough to describe the problem? I have the feeling that it might be a bit complicated to explain my problem.
So my question is: Do I misunderstand the concept of sync/async and simulation speed etc?

Carla: 0.9.11
Ubuntu 18.04 LTS

glopezdiest, 2021-06-11:
Hey @semmelknoedel. Let me recap if I understand this correctly:
1) You are using ScenarioRunner with some modifications such that the ego vehicle is also controlled by SR, and the ego uses the BasicAgent of the PythonAPI agents.
2) You are running the simulation synchronously (20 'simulation fps' and around 200 real ones), while recording the scene.

Regarding your _VehicleController_, you say that it doesn't scale with the speed. Do you mean that it is incapable of controlling the ego? Or is it another issue?

I've checked and the VehicleController doesn't have any supposition about the simulation running in real time, so it should be fine **as long as it is called once per frame**. This is also seen when running routes with SR, as it can use the [npc_agent.py](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/autoagents/npc_agent.py), which is based on the BasicAgent and can run faster than real time.

semmelknoedel, 2021-06-11:
Hi @glopezdiest ,
regarding your questions:

1.  I'm using a modified ScenarioRunner where I included my own trajectoy follower (based on the agent base class) and an added behavior similar to BasicAgent. It returns speed, and coordiantes for its destination. To transfer that to Carla usable information I use the PIDControler which is also used by the BaicAgent with adjusted parameters. These parameters are dependent on the simulation speed and have to be adjusted accordingly.
2. I'm running the simulation synchronously, where my simulation (without a wait from pygame) is much faster and reaches 200 fps, so it simulates 10 seconds within 1 real second.

I'm no control theory expert, but I would expect, that the input should be the same no matter how fast the simulation is running. Right now, as soon as I run faster than the given fps (by that, I mean it is sync mode and according to the time stamps it is still 20 fps but just speeded up to 200 fps or 10 seconds simulated within 1 second or 10x realtime).
However, I need to adjust the PID parameters as if it where runnning with 200 fps within 1 second.
Basically, right now I have to slow down the simulation artificially and I'm trying to find out if the problem is within my trajectory planner or not.

I hope this was clear enough. Thanks a lot for your time.

semmelknoedel, 2021-06-11:
The controller is called in each update() step of the behavior. So I hope it is only called once per frame.
By failing I mean the lateral control is oversteering and usually turing right or left immediatly. Most probably the longitudinal control is also overdoing, but this is harder to see.

glopezdiest, 2021-06-11:
Then, I really don't know what could be happening. Is it possible for you to send me your modified ScenarioRunner code through here? I'll take a look at your ego controller see where the error may be

semmelknoedel, 2021-06-29:
I'm sorry. I can't share the code (part of our control it is running in a docker container where I don't even have it).
I am no control theory expert, but maybe our ego control is the problem. I have to check whether it can handle speeding up the time. Maybe there is a problem by using it together with a framework running faster than realtime.
Since I know now that Carla and the PIDController is working with different code, I'd guess the problem is on our side.
Thanks a lot!

closed 2023-06-02 by stale[bot]