carla-simulator/carla #4266: get_elapsed_time get zero when traffic light state is red

xujiahao97, 2021-06-02:
Hello, i am now using carla.Trafficlight class. i use get_red_light function and it returns 2,but actually red light red light shines 32 seconds. and i use get_elapsed_time function and its return value is correct during yellow light and green light.but it only counts first 2 seconds of red lights and then it returns 0.
I am wondering whether anyone run into same issues or does anyone know how to figure it out.

corkyw10, 2021-06-08:
Hi @xujiahao97, could you provide a code example to help reproduce your issue?

xujiahao97, 2021-06-09:
[GLOSA.txt](https://github.com/carla-simulator/carla/files/6620523/GLOSA.txt)

i wrote this python file according to automatic_control.py.

zdx19981006, 2021-06-10:
meet the same issue

GordenZhao, 2021-08-17:
Hi @corkyw10, is there any progress for this issue reproducing or fixing? I met the same issue. Thank you.

flyico, 2022-09-15:
I got the same problem, anyone has some tips?

closed 2023-06-02 by stale[bot]

kmeeey, 2023-09-21:
This problem still exists in 0.9.14, does anyone have a better way to get the remaining time of the signal light?

smart-geng, 2024-07-08:
This problem still exists in 0.9.15