carla-simulator/carla #4273: Traffic manager Hybrid physics mode is neither producing the colored boxes over Ego vehicle nor disabling the physics

vikas-sutrala, 2021-06-03:
#!/usr/bin/env python

# Copyright (c) 2019 Computer Vision Center (CVC) at the Universitat Autonoma de
# Barcelona (UAB).
#
# This work is licensed under the terms of the MIT license.
# For a copy, see <https://opensource.org/licenses/MIT>.

"""Spawn NPCs into the simulation"""

import glob
import os
import sys
import time

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

from carla import VehicleLightState as vls

import argparse
import logging
from numpy import random

def main():
    argparser = argparse.ArgumentParser(
        description=__doc__)
    argparser.add_argument(
        '--host',
        metavar='H',
        default='127.0.0.1',
        help='IP of the host server (default: 127.0.0.1)')
    argparser.add_argument(
        '-p', '--port',
        metavar='P',
        default=2000,
        type=int,
        help='TCP port to listen to (default: 2000)')
    argparser.add_argument(
        '-n', '--number-of-vehicles',
        metavar='N',
        default=10,
        type=int,
        help='number of vehicles (default: 10)')
    argparser.add_argument(
        '-w', '--number-of-walkers',
        metavar='W',
        default=50,
        type=int,
        help='number of walkers (default: 50)')
    argparser.add_argument(
        '--safe',
        action='store_true',
        help='avoid spawning vehicles prone to accidents')
    argparser.add_argument(
        '--filterv',
        metavar='PATTERN',
        default='vehicle.*',
        help='vehicles filter (default: "vehicle.*")')
    argparser.add_argument(
        '--filterw',
        metavar='PATTERN',
        default='walker.pedestrian.*',
[... 240 lines cut ...]
            settings = world.get_settings()
            settings.synchronous_mode = False
            settings.fixed_delta_seconds = None
            world.apply_settings(settings)

        print('\ndestroying %d vehicles' % len(vehicles_list))
        client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list])

        # stop walker controllers (list is [controller, actor, controller, actor ...])
        for i in range(0, len(all_id), 2):
            all_actors[i].stop()

        print('\ndestroying %d walkers' % len(walkers_list))
        client.apply_batch([carla.command.DestroyActor(x) for x in all_id])

        time.sleep(0.5)

if __name__ == '__main__':

    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')

vikas-sutrala, 2021-06-04:
I opened this issue because I would like to use the Hybrid physics mode feature from CARLA.
as stated in the Carla Traffic manager Hybrid physics mode when I use the spawn_npc.py with hybrid mode enabled, I could not see any blocks on a ego vehicle or near by vehicles in the mentioned radius.

I have also verified the vehicle physics by using the parameter get_physics_control() to check the parameters like mass, use_gear_autobox etc. there were no changes either by setting an ego vehicle or not setting the ego vehicle.
as mentioned in the documentation when there is no ego vehicle and hybrid physics mode is on the vehicle physics should be disable for all the vehicles but that is not the case.
I even tried with synchronous mode enabled.
here is the code that I am using.
there were no changes made to spawn_npc.py except           
                if n==1:
                blueprint.set_attribute('role_name', 'hero')

I am using carla version 0.9.11 and windows 10.

jackbart94, 2021-06-07:
The blocks were a mere tool to show which cars were affected by the Hybrid Physics mode in our release video, they are not included in the feature. If you want to recreate it, you should use the CARLA Debug Helper, there's enough information about it in the PythonAPI.

Also, if you are doing get_physics_control() to get the mass, this will still return the mass of the vehicle. A good method to check if a vehicle has physics activated or not is to use ``get_velocity()``. If it returns 0, either the vehicle is stopped or it has no physics.

On a side note, we've recently found a bug where if the hero vehicle is registered to the Traffic Manager, its physics will be disabled too. This is already fixed and will be soon pushed to ``dev``.

closed 2021-11-17 by germanros1987