carla-simulator/carla #4280: RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 192.168.245.2:2000

lzhhh93, 2021-06-05:
Hi everyone,

Since the old same issue seems to be closed, I create a new one:

I run Carla in my host and launch my ros/C++ code in my guest virtual machine, which communicates with Carla. 

I have followed the advices under the old same issue, but none of that works for me. 
For example, turn on performance mode of my pc, increase the time out parameter from 10 to 200 and press the Play button (or alternatively Alt+P) after launching Unreal Engine, before launching my .launch file. 

The spec of my pc is:  AMD Ryzen 5 5600H/16G RAM(8G for my VM)/RTX 3060.

I have googled for two days but still got nothing, really need some help here.

Thanks in advance.

corkyw10, 2021-06-08:
Hello @lzhhh93, if you are referencing an older issue, could you please link to it here?

Could you also outline the steps you are taking so that we can reproduce your issue? Also include the versions of carla you are working with?

QianwenZhu, 2021-06-28:
for reference:
1. make sure your virtual machine can connect to your host (ping your host's IP address))
2. make sure your carla version is the same.
3. make sure the python version you run client and server is the same.

closed 2023-06-02 by stale[bot]