carla-simulator/carla #4282: unexpected frame number from get_snapshot after apply_batch_sync

qhaas, 2021-06-06:
Possibly related to Issue #2812

On CARLA 0.9.11 in GNU/Linux, either [`apply_batch_sync`](https://carla.readthedocs.io/en/0.9.11/python_api/#carla.WorldSnapshot) or [`get_snapshot`](https://carla.readthedocs.io/en/0.9.11/python_api/#methods_7) appears asynchronous.  Consider:

```
#!/bin/python3

import carla
import traceback
import random
try:
    random.seed(0)

    # setup client/server
    client = carla.Client('localhost', 2000)
    world = client.load_world('Town01')
    settings = world.get_settings()
    settings.fixed_delta_seconds = 1./20
    settings.synchronous_mode = True
    world.apply_settings(settings)
    world.tick()

    # spawn actors
    actors = []
    bpLibrary = world.get_blueprint_library()
    # vehicle
    bp = bpLibrary.filter('vehicle.chevrolet.impala')[0]
    bp.set_attribute('role_name', 'hero')
    transform = world.get_map().get_spawn_points()[5]
    vehicle = world.spawn_actor(bp, transform)
    actors.append(vehicle)
    world.tick()
    
    beforeApplyBatchTick = world.get_snapshot().frame
    commands = []
    commands.append(carla.command.ApplyVehicleControl(vehicle.id, carla.VehicleControl(throttle=random.random(), steer=random.random())))
    results = client.apply_batch_sync(commands, True)
    afterApplyBatchTick = world.get_snapshot().frame
    print('tick before apply_batch_sync {}, tick after apply_batch_sync {}'.format(beforeApplyBatchTick, afterApplyBatchTick))

    print('cleaning up')
    client.apply_batch([carla.command.DestroyActor(x) for x in actors])
    settings.fixed_delta_seconds = None
    settings.synchronous_mode = False
    world.apply_settings(settings)
    world.wait_for_tick()

except Exception as err:
    print(traceback.format_exc())
```

I expect: `tick before apply_batch_sync 7, tick after apply_batch_sync 8`
The result is actually: `tick before apply_batch_sync 7, tick after apply_batch_sync 7`

DSantosO, 2021-06-11:
Hello @qhaas,
Thanks for the report. The tick was happening but the client was not receiving the information in time. We have uploaded a fix for that and it will be in `dev` soon. 
In the meantime, you can just tick from outside the `apply_sync_batch`:
```
    beforeApplyBatchTick = world.get_snapshot().frame
    commands = []
    commands.append(carla.command.ApplyVehicleControl(vehicle.id, carla.VehicleControl(throttle=random.random(), steer=random.random())))
    results = client.apply_batch_sync(commands)
    world.tick()
    afterApplyBatchTick = world.get_snapshot().frame
    print('tick before apply_batch_sync {}, tick after apply_batch_sync {}'.format(beforeApplyBatchTick, afterApplyBatchTick))
```
and everything should work as expected.

closed 2021-07-14 by DSantosO

harshitha0698, 2024-12-18:
image_queue = Queue()
camera.listen(image_queue.put)
frame_count = 0

try:
    while frame_count < 20:  # Process 20 frames
        world.tick()

        # Get RGB and segmentation images:
        if not image_queue.empty() 
            image = image_queue.get()
              **if (image.frame) == (world.get_snapshot().timestamp.frame):**
                      image.save_to_disk(os.path.join(rgb_images)
              
     if (image.frame) == (world.get_snapshot().timestamp.frame):  
Here, the condition is always evaluating to False because the values of world.frame and image.frame are not synchronized correctly. Specifically, it seems that world.frame is consistently higher than image.frame. 
Can anyone explain this situation