carla-simulator/carla #4286: I'd want to specify a path with autopilot.

hi6724, 2021-06-08:
I want to specify a path with autopilot.
how can I do it?
and I want to know specify a path without autopilot too

qhaas, 2021-06-10:
Not sure you can with autopilot / trafficmanager, but the undocumented(?) [agent API](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/basic_agent.py#L47) might handle this.

glopezdiest, 2021-06-16:
Hey @hi6724, what @qhaas is exactly right. Adding to the comment, the autopilot doesn't currently allow to specify a path for vehicles, choosign randomly one diretion when faced with multiple paths. We have plans to do that but as of right now, you can't.

Regarding the [agent API](https://github.com/carla-simulator/carla/tree/0.9.11/PythonAPI/carla/agents/navigation), it does indeed allow you to do so. It is currently undocummented, so here is a bit more detail on how it works. By using the [basic agent](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/basic_agent.py) you can [set a destination](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/basic_agent.py#L47) of an actor.  This is done using the [global route planner](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/global_route_planner.py), which creates a graph and gets the shortest route, as well as the [local planner](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/carla/agents/navigation/local_planner.py), which uses PIDs to control the vehicle.

LucasFKobernic, 2021-06-17:
Hi, 

I'm using carla 0.9.11 in Windows 10. I tried to use the agent API like u recommend before. But i getting the error 

`name 'BasicAgent' is not defined`

What make no sense, sice the other function from PythonAPI work fine. Have you any idea, what the problem is?

Thanks!

glopezdiest, 2021-11-24:
As of 0.9.13, the TrafficManager now allows setting paths for its vehicles. I don't think we have the propert documetnation ready yet, so you can check [this PR](https://github.com/carla-simulator/carla/pull/4786), which has all the new features explained

closed 2021-11-24 by glopezdiest