carla-simulator/carla #4288: No CarlaUE4.exe in main folder after installation (Windows10)

NicoV1994, 2021-06-08:
Hey maby thats a stupid question but im just getting startet.
Im using Windows 10 and so far everything works. I can start the simulation and execute python scripts.

My question is: In the quickstart guid (https://carla.readthedocs.io/en/0.9.11/start_quickstart/) it says that you can start Carla from the main carla folder with CarlaUE4.exe . But i don't have any executable in the main folder. Did i do somethin wrong?

I have an exe in carla\Unreal\CarlaUE4\Binaries\Win64\CarlaUE4.exe but i can't start from there because it says its "uncooked". Whatever that means.
I can start with carla\Unreal\CarlaUE4CarlaUE4.uproject. But i can't choos a map with the terminal.

But i just want to start carla server from the terminal with the commands to choose a map and so on.
Please help me.

NicoV1994, 2021-06-08:
To be more specific. I can't start Carla in it's own window. I can only start it in the Unreal Editor

corkyw10, 2021-06-08:
Hello @NicoV1994, it sounds like you are working with the source repo of CARLA. Did you download the binary release from [here](https://github.com/carla-simulator/carla/releases/tag/0.9.11)? The quickstart refers to the packaged/binary release of CARLA.

NicoV1994, 2021-06-08:
Sorry i didn't want to confuse. I installed it like in this guide https://carla.readthedocs.io/en/0.9.11/build_windows/
So i pulled it from git

corkyw10, 2021-06-09:
The one you pulled from githun needs to be compiled according to the instructions in that [document](https://carla.readthedocs.io/en/0.9.11/build_windows/). If you want the executable (CarlaUE4.exe) as it mentions in the quickstart. you should use the package version of Carla I mentioned [here](https://github.com/carla-simulator/carla/releases/tag/0.9.11)

NicoV1994, 2021-07-14:
Wasn't online for some time due to illness.
I now used the package version of carla and i can run it with ".\CarlaUE4.exe" (not with CarlaUE4.exe like in the guide https://carla.readthedocs.io/en/0.9.11/start_quickstart/#running-carla).

But it seems like i don't get it. If i run ".\CarlaUE4.exe /Game/Maps/Town02" it alwais launches with Town01. 
Maby the guide is false, but i don't know what to do different.

Hope someone can help a beginner

corkyw10, 2021-07-14:
I'm not sure where you are seeing in that guide to load a map that way. The guide says to use `config.py`;

`> ./config.py --map Town05        # Change map`

corkyw10, 2021-08-05:
I'm closing this issue as I am leaving the CARLA team. I think we have covered your questions so far. If anything new arises, please open a new github issue. Thank you!

closed 2021-08-05 by corkyw10