carla-simulator/carla #4289: Fps lower after adding players

zjh6252007, 2021-06-08:
Hello everyone,
Now I'm doing a multiple driving test. I have a computer as server and another three as clients. When I connect with one player, the server fps down from 70 to 30. When I connect with two players, the fps down from 30-20. And with three players, the server fps down to 15. I guess that the increase in RGB cameras has led to a decrease in FPS. Is there any way I can improve the fps?

qhaas, 2021-06-09:
According to the [carla performance report](https://carla.readthedocs.io/en/latest/adv_benchmarking/#carla-performance-report), adding vehicles does decrease performance significantly.  If they are managed by autopilot (traffic manager), performance decreases further, in part because [TM appears to be single-threaded](https://github.com/carla-simulator/carla/issues/3442#issuecomment-846634130) (cpu bound).

You can check to see if you are CPU or GPU bound by using `nvidia-smi` and `top` to compare GPU vs CPU if you are in no mood to profile the code.  If you see the python process pegged at 100% CPU, likely means you have a bottleneck with some feature that is single-threaded.

If you are GPU bound, [graphics performance tweaks](https://carla.readthedocs.io/en/latest/adv_rendering_options/#graphics-quality) might not help.

If you are CPU bound...  enabling [hybrid mode](https://carla.readthedocs.io/en/latest/adv_traffic_manager/#hybrid-physics-mode) can help with physics induced coy bottlenecks, depending on how many NPCs are in the simulation that are not near your players.  If you are saving imagery out as PNGs, [know libpng is single-threaded](https://github.com/carla-simulator/carla/issues/3895), but one can switch to tiff to mitigate.

Also, if you are using [synchronous mode](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/) with multiple clients, then you might be in uncharted performance territory ;)

liuyandong1988, 2021-07-21:
> According to the [carla performance report](https://carla.readthedocs.io/en/latest/adv_benchmarking/#carla-performance-report), adding vehicles does decrease performance significantly. If they are managed by autopilot (traffic manager), performance decreases further, in part because [TM appears to be single-threaded](https://github.com/carla-simulator/carla/issues/3442#issuecomment-846634130) (cpu bound).
> 
> You can check to see if you are GPU or CPU bound by using `nvidia-smi` and `top` to compare GPU vs CPU if you are in no mood to profile the code. If you see the python process pegged at 100% CPU, likely means you have a bottleneck with some feature that is single-threaded.
> 
> If you are GPU bound, [graphics performance tweaks](https://carla.readthedocs.io/en/latest/adv_rendering_options/#graphics-quality) might help.
> 
> If you are CPU bound... enabling [hybrid mode](https://carla.readthedocs.io/en/latest/adv_traffic_manager/#hybrid-physics-mode) can help with physics induced coy bottlenecks, depending on how many NPCs are in the simulation that are not near your players. If you are saving imagery out as PNGs, [know libpng is single-threaded](https://github.com/carla-simulator/carla/issues/3895), but one can switch to tiff to mitigate.
> 
> Also, if you are using [synchronous mode](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/) with multiple clients, then you might be in uncharted performance territory ;)

I try the _hybrid mode_  method(fig 1). In the beginning, the fps is almost 10 fps(fig 2). However, the FPS also decreases after some episodes (fig 3). I use the synchronous mode, and server/client fps is 10. I set 50 vehicles and 50 pedestrains in Town02 (Carla 0.9.11). I have to restart my program script. This problem has bothered me for a long time.

[image]
[image]
[image]

closed 2023-06-02 by stale[bot]