carla-simulator/carla #4291: Walkers stopping at Curb

ArunaachalamMuralidharan, 2021-06-10:
Hello Everyone,
I am trying to control the Walkers via WalkerControl(), and the walkers sometime stops at the curb when moving from road to sidewalk. (I have attached the picture where the Walker stops)
[image]
 Similarly, at times, the walkers do not climb to the sidewalk but stride along the curbs, as if obstructed by an obstacle.

I have found a fix, which could be done in the UE4 Walker blueprint, by changing the parameters for Character Movement. I am looking for a way to change the Walker Blueprint variables via PythonAPI. Is there any possibility available?

Or is there any other possible options to stop this from happening?

Thanks again.

CARLA 0.9.11
Ubuntu 18.04

erosinnocenti, 2022-02-10:
Hello everyone, same problem here.
Did you find a solution?
Thank you!

andrewbest-tri, 2024-07-11:
This is still occurring in Town15. Please re-open oh stalebot