carla-simulator/carla #4311: Randomness of trees for different simulation sessions

ttthl, 2021-06-14:
Hello,

I found that the trees and plants in the same map could be different when loading the world for several times. My goal is to have same scenarios for two running times with two different car models (the only difference is the material of the car), so I followed the structure of the example `/test/smoke/test_determinism.py` to get deterministic results of the total traffic and add camera sensor on one ego car to retrieve images. However, even when the location of the car is exactly the same for two set of image sequences, sometimes the shapes of the trees might have differences.

Here are several image pair examples.
[image] [image]
You can see that due to the differences of the trees, the shadows on the building will also change.

[image] [image]
Here the distribution of the trees on the small mountain also changes.

[image] [image]
And the plants alongside the road can also be quite different.

So I'm wondering if there is any way to have the same generation of all trees and plants for each simulation? 

Implementation details:
Carla version: 0.9.11
Ubuntu 18.04

Thank you

XGodina, 2021-06-15:
Hi @ttthl  The thing that causes that is the wind. Our Trees move depending on the strength of the wind. Use our script environment.py to reduce the strength of wind to 0.

ttthl, 2021-06-15:
Hi @XGodina , thanks for the reply. However I've already set the strength of wind to 0. Is there any other reason can cause this?

XGodina, 2021-06-30:
Hi, @ttthl 
First of all forgiveness for answering so late. Yes, @DSantosO discovered that determinism did not work well.
I guess this is what you're talking about. But I think he will explain better than me, what was the problem.

XGodina, 2021-07-19:
Hi, @ttthl We fix this problem. The next release will be in a couple of weeks.

ttthl, 2021-07-19:
Hi, @XGodina Thanks for the remind. That would be great, I'll try it for the next release.

closed 2022-04-29 by stale[bot]