carla-simulator/carla #4314: Runtime Timeout Error in Spawn_npc.py script after adding New map

HarishKumar051993, 2021-06-16:
Hi All,

I have created a new map using RoadRunner and currently using Carla 9.11 version.

After creating a map, when i try to run the Spawn_npc.py script, Iam getting a Runtime error as below:

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.11-5-g4c8f4d5f 
WARNING: Simulator API version  = 0.9.11-5-g4c8f4d5f-dirty 
WARNING: cannot parse georeference: ''. Using default values. 

destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 303, in <module>
    main()
  File "spawn_npc.py", line 144, in main
    spawn_points = world.get_map().get_spawn_points()
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

The above error i am facing only in **LINUX** (Not in Windows). And all other scripts as Automaticcontrol.py and manual control.py is working fine in Linux machine.
This Error iam getting only to the new map which i created. In other Town Maps, the Spawn_npc.py script is working.

Can anyone help why iam facing this error.

jackbart94, 2021-06-17:
Try increasing the timeout first, especially if the map is quite big. You must change the value in this line:

`` client.set_timeout(10.0) ``

Otherwise, it might be a different problem.

HarishKumar051993, 2021-06-17:
> Try increasing the timeout first, especially if the map is quite big. You must change the value in this line:
> 
> `client.set_timeout(10.0)`
> 
> Otherwise, it might be a different problem.

@jackbart94 Yeah i tried increasing the time as well. Still the same issue.

HarishKumar051993, 2021-06-17:
@jackbart94  I have created a map without Pedestrians and Traffic signals. Is that be a problem?

jackbart94, 2021-06-17:
> @jackbart94 I have created a map without Pedestrians and Traffic signals. Is that be a problem?

Try following carefully the documentation on how to create a map. It appears that spawn points are the problem here.

HarishKumar051993, 2021-06-17:
> > @jackbart94 I have created a map without Pedestrians and Traffic signals. Is that be a problem?
> 
> Try following carefully the documentation on how to create a map. It appears that spawn points are the problem here.

Yeah, as per the document The vehicle spawn points are added through the Opendrive Actor and even I got 47 Vehicle spawn points in it.

The same works fine for Windows. Only in Linux i could not able to spawn

HarishKumar051993, 2021-06-18:
> > @jackbart94 I have created a map without Pedestrians and Traffic signals. Is that be a problem?
> 
> Try following carefully the documentation on how to create a map. It appears that spawn points are the problem here.

Adding one more observation as, the spawn_npc script is working in no rendering mode perfectly.
Only it is not running when i play the map in unreal Editor and trying to spawn in the map.

jackbart94, 2021-06-18:
It has to be the timeout. There's no other issue I can think of. Could you try and increase it significantly, like 60 seconds or even 120, then eventually paste here the error?

HarishKumar051993, 2021-06-18:
> It has to be the timeout. There's no other issue I can think of. Could you try and increase it significantly, like 60 seconds or even 120, then eventually paste here the error?

Still I got the same error as below:

destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 306, in <module>
    main()
  File "spawn_npc.py", line 146, in main
    spawn_points = world.get_map().get_spawn_points()
RuntimeError: time-out of 120000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

jackbart94, 2021-06-18:
But ``manual_control.py`` has the same exact call and you're saying it works, so I'm a bit clueless honestly. Is the simulator running correctly before you run ``spawn_npc.py``? Which version of CARLA are you using?

HarishKumar051993, 2021-06-18:
> But `manual_control.py` has the same exact call and you're saying it works, so I'm a bit clueless honestly. Is the simulator running correctly before you run `spawn_npc.py`? Which version of CARLA are you using?

Yeah the simulator was running through Unreal Editor(I played the Game from unreal Editor) and i ran the spawn_npc.py

Iam using carla version 9.11

Ria9929, 2021-06-20:
make sure your port 2000 and 2001 is open and accessible and no app is blocking it .

HarishKumar051993, 2021-06-21:
Signal 11 caught. Malloc Size=65538 LargeMemoryPoolOffset=65554  CommonUnixCrashHandler: Signal=11 Malloc Size=65535 LargeMemoryPoolOffset=131119  Malloc Size=145408 LargeMemoryPoolOffset=276544  Engine crash handling finished; re-raising signal 11 for the default handler. Good bye. Segmentation fault (core dumped)

I get the above error on running Spawn_npc.py in **Carla Server Build (from source)** using Docker

HarishKumar051993, 2021-06-21:
> make sure your port 2000 and 2001 is open and accessible and no app is blocking it .

@Ria9929 Yeah there is no problem on that

Ria9929, 2021-06-22:
see if this  might help 
https://github.com/carla-simulator/carla/issues/3193

HarishKumar051993, 2021-06-24:
> see if this might help
> #3193

@Ria9929 Thanks. I went through the mentioned issue, but still i am facing same issue.

jackbart94, 2021-11-18:
Try with a newest version of CARLA, see if it's still happening.

> Signal 11 caught. Malloc Size=65538 LargeMemoryPoolOffset=65554 CommonUnixCrashHandler: Signal=11 Malloc Size=65535 LargeMemoryPoolOffset=131119 Malloc Size=145408 LargeMemoryPoolOffset=276544 [...]

This means you have no memory available.

closed 2021-11-18 by jackbart94