carla-simulator/carla #4318: [Installation Problems - Windowns] - fatal error C1083:  "carla/Memory.h": No such file or directory

LucasFKobernic, 2021-06-18:
Hi everyone, 

I have already installed CARLA 0.9.11 in my Windows 10 Computer. But since i have restaterd my Computer yerstady i'm having problems to start Carla. 

I'm getting the following error, when i try it:
[image]

When i choose to rebuild i got this error:
[image]

The same happens when a tried  make launch.

By using make rebuild i got the following error:

```
Verstrichene Zeit 00:00:00.92
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\Users\z239023\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --rebuild
-[BuildPythonAPI]: Cleaning "C:\Users\z239023\carla\PythonAPI\carla\build"
-[BuildPythonAPI]: Cleaning "C:\Users\z239023\carla\PythonAPI\carla\source\carla.egg-info"
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
creating source\carla.egg-info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib.win-amd64-3.7
creating build\lib.win-amd64-3.7\carla
copying source\carla\command.py -> build\lib.win-amd64-3.7\carla
copying source\carla\__init__.py -> build\lib.win-amd64-3.7\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-3.7
creating build\temp.win-amd64-3.7\Release
creating build\temp.win-amd64-3.7\Release\source
creating build\temp.win-amd64-3.7\Release\source\libcarla
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include "-IC:\Program Files\Python37\include" "-IC:\Program Files\Python37\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
libcarla.cpp
source/libcarla/libcarla.cpp(7): fatal error C1083: Datei (Include) kann nicht geöffnet werden: "carla/Memory.h": No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\z239023\carla\PythonAPI\carla\dist"!
```

Has anyone faced this problem? Or do have any idea how to solve it? 

The interesting thing is that before restarting the computer Carla was running.

LucasFKobernic, 2021-06-18:
I have re installed UE4 and now i can open CARLA. But by trying to run any python script i got the following error:

```
Traceback (most recent call last):
  File "spawn_npc.py", line 24, in <module>
    import carla
ModuleNotFoundError: No module named 'carla'
```

Any idea?

LucasFKobernic, 2021-06-18:
I run make pythonAPI again now it works.

closed 2021-06-18 by LucasFKobernic

Lijingqi97, 2023-03-10:
> Hi everyone,
>
> I have already installed CARLA 0.9.11 in my Windows 10 Computer. But since i have restaterd my Computer yerstady i'm having problems to start Carla.
>
> I'm getting the following error, [...]

> 

hello, I encounter the same problem that 'carla/Memory.h' can not be found, how did you solve this problem?

zyvonne, 2023-10-22:
> hello, I encounter the same problem that 'carla/Memory.h' can not be found, how did you solve this problem?

same issue, did you solve this problem?

CXLI1124, 2025-02-23:
Don't worry about it, just run make launch. After my testing, it can be used normally. In subsequent use, simply run the version corresponding to pip install carla to run it

hk1152g, 2025-04-15:
> Don't worry about it, just run make launch. After my testing, it can be used normally. In subsequent use, simply run the version corresponding to pip install carla to run it

Hello, I'm not sure I understand. So I can just ignore the error concerning carla/Memory.h? What does this mean "In subsequent use, simply run the version corresponding to pip install carla to run it"?