carla-simulator/carla #4320: Problem with "BehaviorAgent"

LucasFKobernic, 2021-06-18:
Hi everyone, 

I have being testing BasicAgent, RoamingAgent and BehaviorAgent in the last view days. 

At first a tested it with automatic_control.py. Then i wrote my own python script. I had no problem with BasicAgent and with RoamingAgent. But when i try to apply a BehaviorAgent i got the following error:

`Traceback (most recent call last):
  File "Scenario_4.py", line 217, in <module>
    main()
  File "Scenario_4.py", line 185, in main
    driving_vehicle.apply_control(agent.run_step())
  File "C:\Users\z239023\carla\PythonAPI/carla\agents\navigation\behavior_agent.py", line 415, in run_step
    elif self.incoming_waypoint.is_junction and (self.incoming_direction == RoadOption.LEFT or self.incoming_direction == RoadOption.RIGHT):
AttributeError: 'NoneType' object has no attribute 'is_junction'`

Am i using BehaviorModel wrong ? Or  where is the problem ?

Thanks!

kar-ab, 2021-06-21:
Hello @LucasFKobernic,

If you go through automatic_control.py carefully, only for BehaviorAgent, a member function `agent.update_information()` is called before `agent.run_step()`. This call is necessary to update the ego vehicle based on surroundings, add following line as the first line of the `while True` loop. 
```
agent.update_information(world)
```
Additionally, make sure that 'world' object has attribute 'player' set as the vehicle to be controlled.
If not, you can make use of below class and change the update_info call:
```
class World(object):
	    def __init__(self, carla_world, ego_vehicle):
		self.world = l
		self.player = ego_vehicle

agent.update_information(World(world, ego_vehicle))
```

LucasFKobernic, 2021-06-25:
Hi @kar-ab , 

thanks for the answer. I tried your suggestion and it works!

closed 2021-06-25 by LucasFKobernic