carla-simulator/carla #4327: fatal error C1083 with make PythonAPI command

vibhoraggarwal, 2021-06-21:
#### Problem description
I am building Carla in Windows from source. While running `make PythonAPI` command I get following error: `fatal error C1083: Cannot open include file: 'recast/Recast.h': No such file or directory` and the `.egg` file does not get created in `C:\git\carla\PythonAPI\carla\dist`

#### Things already done and versions
- Installing using the information on this link: https://github.com/carla-simulator/carla/blob/master/Docs/build_windows.md
-    | Software| Version|Remark
      | ----------- | ----------- |-----------|
      | Operating system| Windows 10|-|
      | Visual Studio| 2017|-|
      | UnrealEngine| 4.24|The unreal engine works fine using the `UnrealEngine\Engine\Binaries\Win64\UE4Editor.exe`|
- I also got the same error with `boost/...` files but then I included the boost directory `C:\git\carla\Build\boost-1.72.0-install\include` as the `INCLUDE` system environment variable, and it was solved. However, it seems like never ending inclusion of all the include folders manually, creating a doubt as to something is wrong.

#### Similar issues/links
1. Issue #3993 also mentions a similar point. However, it is not solved as well. So, I just created another issue for better visibility and some steps that I already did.

#### Question

1. Can somebody also reproduce the same error while running the command `make PythonAPI` as well?
2. There was no clear understanding as to how the locations were provided during the `make PythonAPI` command? 

#### Log
```
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "C:\git\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\git\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\git\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\git\carla\Build\" --zlib-install-dir "C:\git\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\git\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\git\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\git\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\git\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\git\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\git\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\git\carla\Build\recast-cdce4e-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\git\carla\Build\" --toolset msvc-14.1 --version 1.72.0 -j 8
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\git\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\git\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\git\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!
[... 64 lines cut ...]
  -- Install configuration: "Release"
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\git\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include "-IC:\Program Files\Python38\include" "-IC:\Program Files\Python38\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" -IC:\git\FMI-Standard-2.0.1\headers "-IC:\Program Files\Python38\include" -IC:\git\protobuf-2.6.1\src -IC:\git\open-simulation-interface\build -IC:\git\carla\LibCarla\source -IC:\git\vcpkg\installed\x86-windows\include\boost -IC:\git\carla\Build\boost-1.72.0-install\include -IC:\git\carla\Build\rpclib-install\include /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.8\Release\source/libcarla/libcarla.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
libcarla.cpp
C:\git\carla\LibCarla\source\carla/nav/Navigation.h(16): fatal error C1083: Cannot open include file: 'recast/Recast.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\git\carla\PythonAPI\carla\dist"!
```

LucasFKobernic, 2021-07-26:
Hi, i'm facing a similar error. Did you solved it ?

vibhoraggarwal, 2021-07-27:
Hi @LucasFKobernic 

No, I haven't solved it yet. I am waiting for some answers from the developers here.

vibhoraggarwal, 2021-07-27:
@LucasFKobernic Did you also try it on some other Visual Studio configurations?

LucasFKobernic, 2021-07-28:
Hi, 

Which CARLA version are you trying to install?

I found that for the dev version need to be installed with VisualStudio 2019 and not 2017. ( They have change it recently and isn't the documentation yet).

I installed the version 9.11 with Visual Studio 2017 and it worked!

vibhoraggarwal, 2021-07-28:
I was installing v0.9.11 as well.

closed 2023-06-02 by stale[bot]