carla-simulator/carla #4328: The semantic segmentation image has lots of black points/pixels

JinRanYAO, 2021-06-22:
Hello, I save the semantic segmentation image to the disk. but the image has lots of black points/pixels, for example:
[image]
Could you tell me how to fix it?
It is the code that I save the semantic segmentation image to the disk:
`blueprint_library = client.get_world().get_blueprint_library()
camera_bp = blueprint_library.find('sensor.camera.semantic_segmentation')
camera_transform = carla.Transform(carla.Location(x=1.5, z=2.4))
camera = client.get_world().spawn_actor(camera_bp, camera_transform, attach_to=world.player)

cc = carla.ColorConverter.CityScapesPalette
camera.listen(lambda image: image.save_to_disk('segmentation_image/%06d.png' % image.frame, cc))`
Look forward to your reply!

qhaas, 2021-06-22:
1. Which version of CARLA?
2. Do artifacts appear in the same place with an RGB camera image?
3. Did you try a [different renderer](https://carla.readthedocs.io/en/0.9.11/adv_rendering_options/#vulkan-vs-opengl)?

JinRanYAO, 2021-06-22:
> Possibly related to Issue #3865 , especially with [newer GPUs](https://github.com/carla-simulator/carla/issues/3865#issuecomment-777357195)
> 
>     1. Which version of CARLA?
> 
>     2. Do artifacts appear in the same place with an RGB camera image?
> 
>     3. Did you try a [different renderer](https://carla.readthedocs.io/en/0.9.11/adv_rendering_options/#vulkan-vs-opengl)?

Hello, thank you for your reply.
It's indeed very similar to Issue #3865 , And I use a new GPU RTX 3090.
1. I use CARLA 0.9.11, UnrealEngine 4.24.
2. It happens with a raw RGB camera image too, but it looks like there are fewer black artifacts.
3. I try rendering and no rendering, it happens both the settings,
So is it a bug with RTX 3090? Could I fix it now and how could I fix it?
Look forward to your reply!

qhaas, 2021-06-22:
All my GPUs are too old for this issue to manifest, so I can't say for certain!  But, if they think it is an UnrealEngine glitch and not a CARLA glitch, you might want to ask over at the [unrealengine answershub](https://answers.unrealengine.com/index.html) or attempt to compile from source using the [unofficial CARLA 0.9.11 patch](https://github.com/carla-simulator/carla/issues/3214#issuecomment-769140155) that makes it work with UE 4.25.x (unclear if updating to UE 4.25.x is enough since the bug report only reports a fix in UE 4.26.x).

JinRanYAO, 2021-06-23:
> All my GPUs are too old for this issue to manifest, so I can't say for certain! But, if they think it is an UnrealEngine glitch and not a CARLA glitch, you might want to ask over at the [...]

OK, I will try it, thank you for your suggestion!

lookingworld, 2021-06-28:
You can try adding '-opengl' to the startup command

JinRanYAO, 2021-06-28:
> You can try adding '-opengl' to the startup command

Thank you for your reply, I will try it!

qhaas, 2021-06-28:
> You can try adding '-opengl' to the startup command

That has its own issues in certain weather conditions, see Issue #3377

DSantosO, 2021-08-04:
Hello @JinRanYAO,
Sorry for the late response. I have not been able to reproduce this. Can you test it in the new [0.9.12](http://carla.org/2021/08/02/release-0.9.12/) to check if it is still happening?

qhaas, 2021-08-14:
Seeing the following artifacts with a standard RBG camera in CARLA 0.9.12 running headless vulkan, possibly related, investigating the possibility it is an artifact of running on an A100 GPU (driver 460.91.03), will test later with a Tesla V100 GPU.
[image]

lookingworld, 2022-01-09:
您好     您的邮件我已收到，感谢！祝好

ll7, 2022-02-25:
I have similar problems with Carla0.9.10.1 in Ubuntu 20.04 and a RTX3060TI (Driver Version: 510.47.03)

[image]

[image]

I will try to test this with Carla0.9.13.

swsamleo, 2022-11-10:
I have a similar issue with A100 +Ubuntu 20.04 I used the Carla 0.9.13

airobotis, 2024-02-05:
I have a similar issue with A100 and Ubuntu 20.04.
I used the Carla 0.9.11

buaazeus, 2024-02-22:
same issue 
GPU:A100 
carla:0.9.10

cuijiaxun, 2024-05-01:
Same Issue
GPU: A100
Carla: 0.9.14
----
Any conclusive solutions yet?